WEC-Sim/WEC-Sim #359: WAMIT BEMIO

IloveJesusmyKing, 2020-06-03:
Hello,

Recently I attempted to use the BEMIO workflow that is provided with WEC-Sim and I have run into some issues. Specifically, hydro data missing following the execution of "Read_WAMIT". This lack of data propagates errors when trying to run "Write_H5" and "Plot_BEMIO". I am wondering if there is documentation available regarding the hydro data structure fields. I am wondering if there are outputs I am missing from my WAMIT run that these procedures are trying to extract.

The data fields I am missing that produced errors are below:

- ex_K
- ex_t
- ex_w
- ra_K
- ra_t
- ra_w
- ra_t

Regards,

Nick

dforbush2, 2020-06-03:
Hi @IloveJesusmyKing , 

Are you running the bemio example, with one of the *.out wamit files in the wecsim distribution, or with your own wamit output file?
In the latter case, there may be something unusual about your wamit output. Looking at the fields you're missing, it Seems all of these variables are related to impulse response functions generated by `Radiation_IRF.m` and `Excitation_IRF.m`. Are you able to run execute these scripts? 

There is an example of the BEMIO workflow for WAMIT outputs: 

WEC-Sim/examples/BEMIO/WAMIT 

that outlines the function order: if you run 'Write_H5.m' and 'Plot_BEMIO.m' before these functions, you'll get errors. At least, I get errors if I try this in the above-mentioned example. Not to say there is not a more complex issue going on, so please upload your *.out file if following the bemio.m example function order does not solve the problem.

-Dom

IloveJesusmyKing, 2020-06-04:
- I am running the scripts while processing my own WAMIT *.out file.
- Thanks for letting me know about the `Radiation_IRF.m` and `Excitation_IRF.m` dependencies. I had these in the bemio workflow but they were commented out because I didn't think they were necessary.

Now, as I run through the bemio workflow (like the example you pointed to), I receive one error related to the Ainf hydro data field. The error arises in `Write_H5.m` on line 64 calling on this field. But when I comment it out, everything runs smoothly. 

I inspected other dependencies a little bit and was thinking that my error may be also tied to `Read_WAMIT.m` line 105. However, I am not sure what this line of code is looking for so I don't know if my *.out file is lacking something.

Attached is my *.out file. Please let me know your thoughts

Nick
[sphere_dn_1_4.zip](https://github.com/WEC-Sim/WEC-Sim/files/4729505/sphere_dn_1_4.zip)

IloveJesusmyKing, 2020-06-10:
> Hi @dforbush2
> 
> * I am running the scripts while processing my own WAMIT *.out file.
> * Thanks for letting me know about the `Radiation_IRF.m` and `Excitation_IRF.m` dependencies. I had these in the bemio workflow but they were commented out because I didn't think they were necessary.
> 
> Now, as I run through the bemio workflow (like the example you pointed to), I receive one error related to the Ainf hydro data field. The error arises in `Write_H5.m` on line 64 calling on this field. But when I comment it out, everything runs smoothly.
> 
> I inspected other dependencies a little bit and was thinking that my error may be also tied to `Read_WAMIT.m` line 105. However, I am not sure what this line of code is looking for so I don't know if my *.out file is lacking something.
> 
> Attached is my *.out file. Please let me know your thoughts
> 
> Nick
> [sphere_dn_1_4.zip](https://github.com/WEC-Sim/WEC-Sim/files/4729505/sphere_dn_1_4.zip)

> Hi @dforbush2
> 
> * I am running the scripts while processing my own WAMIT *.out file.
> * Thanks for letting me know about the `Radiation_IRF.m` and `Excitation_IRF.m` dependencies. I had these in the bemio workflow but they were commented out because I didn't think they were necessary.
> 
> Now, as I run through the bemio workflow (like the example you pointed to), I receive one error related to the Ainf hydro data field. The error arises in `Write_H5.m` on line 64 calling on this field. But when I comment it out, everything runs smoothly.
> 
> I inspected other dependencies a little bit and was thinking that my error may be also tied to `Read_WAMIT.m` line 105. However, I am not sure what this line of code is looking for so I don't know if my *.out file is lacking something.
> 
> Attached is my *.out file. Please let me know your thoughts
> 
> Nick
> [sphere_dn_1_4.zip](https://github.com/WEC-Sim/WEC-Sim/files/4729505/sphere_dn_1_4.zip)

closed 2020-06-10 by IloveJesusmyKing