WEC-Sim/WEC-Sim #362: Modeling Extreme Wave Conditions?

sugarl14, 2020-06-08:
Hi,

I am aware that the WEC-Sim method is not intended to model highly nonlinear hydrodynamic events, such as wave slamming and wave breaking, but I noticed in the publication [here](https://www.nrel.gov/docs/fy15osti/63677.pdf?gathStatIcon=true) , that extreme wave conditions were indeed modeled. 

Is a RANS method still required to analyze extreme wave conditions, or can it now be done in WEC-Sim? If not, can ANSYS Fluent be used for RANS? Are there any tutorials on how to import the time-domain data from WEC-Sim to ANSYS FLUENT?

Thank you!

yuyihsiang, 2020-06-15:
Although WEC-Sim is based on linear theory, it can be used for extreme condition analysis with some caveats. WEC-Sim can not simulate the wave slamming and wave breaking loads. However, if the wave slamming and wave breaking on the physical interests is relatively insignificant, WEC-Sim can be used. In some cases, we also use WEC-Sim to identify the most likely sea states that extreme condition occurs, where WEC-Sim simulation is much quicker compared to CFD simulation. Then, we can use CFD or perform experimental tests for the given sea states for further investigation. WEC-Sim also has the option for weakly nonlinear simulation, where you have to turn on the nonlinear hydro option. The option allows WEC-Sim to include the nonlinear restoring and FK forces, based on the instantaneous wave elevation and body position. This can capture the transition nonlinear-drifting nicely and the influence of overtopping to a certain degree. However, this requires a high-resolution panel on the body surface, and the simulation will be much slower compared to the linear WEC-Sim model. 

For your last question: "Are there any tutorials on how to import the time-domain data from WEC-Sim to ANSYS FLUENT?", the short answer is no. I am not sure what time-domain data you are trying to import into FLUENT. If it is just wave elevation, theoretically, you can do that by converting your wave elevation at the WEC to the wave elevation at the boundary using linear or-order wave theory.

kmruehl, 2020-06-18:
@sugarl14 did this answer your question?

sugarl14, 2020-06-22:
It did, thank you!

closed 2020-06-22 by sugarl14