WEC-Sim/WEC-Sim #363: Flexible membrane WEC

RApmmh, 2020-06-09:
Hello,

I am trying to simulate a flexible membrane WEC, similar to an underwater flag with NEMOH code as input. Having found this article ["Inclusion of Structural Flexibility in Design Load Analysis for Wave Energy Converters"](https://www.nrel.gov/docs/fy17osti/68923.pdf) I thought I could use WEC-Sim for the simulation. However, I have run into some problems with my code:

Whenever I add a "flex body" to my simulink model I get this error (the RM3 and standard examples work fine):
_"Error using wecSim (line 291)
Error due to multiple causes.
Caused by:
    Error using wecSim (line 291)
    Failed to load library 'WECSim_Lib_2015b' referenced by 'moda/Flex Body/Additional Linear Damping Force
    Calculation/Additional Linear Damping Force Calculation'
    Error using wecSim (line 291)
    Failed to load library 'WECSim_Lib_2015b' referenced by 'moda/Flex Body/B2B Subsystem'
    Error using wecSim (line 291)
    Failed to load library 'WECSim_Lib_2015b' referenced by 'moda/Flex Body/Morrison Element and
    ViscousDampingForceCalculation/Morison Element and ViscousDampingForceCalculation'
    Error using wecSim (line 291)
    Failed to load library 'WECSim_Lib_2015b' referenced by 'moda/Flex Body/Nonlinear Wave Elevation'
    Error using wecSim (line 291)
    Failed to load library 'WECSim_Lib_2015b' referenced by 'moda/Flex Body/Wave Radiation Forces Calculation/Wave
    Radiation Forces Calculation' "_

**And besides that, I can't seem to understand clearly how a flexible body can be used in WEC-Sim. Does it work? Can it be the only hydrodynamic body?**

I haven't found any documentation on using WEC-Sim with a flexible body, and no information on the  'flex body'   body element either. Could you provide more info on that?

Thank you

yuyihsiang, 2020-06-15:
The library link in WEC-Sim model was overwritten. It is fixed now, please download the latest version.

RApmmh, 2020-06-16:
Thank you, I've pulled the most recent version with git. The library error is gone but I have a different bug concerning the simulink model: 
_Error using wecSim (line 291)
For dimensions 1, parameter 'Indices' in 'test4_model/Flex Body/Additional Linear Damping Force Calculation/Selector1':
error in 'test4_model/Flex Body/Additional Linear Damping Force Calculation/Selector1'. Parameter 'Indices' containing
expression "[7:body.dof_end]" is invalid. The parameter must evaluate to real integer values_
body.dof_end is equal to 6 so the problem is that this value is not larger than 7. However, the standard number of degrees of freedom obtained using NEMOH is 6. Does this mean that to use a flexible body in WEC-Sim we have to define our own supplementary degrees of freedom in NEMOH (ex. the normal mode deformations of the flexible body)? Or do we have to use another type of input than NEMOH to be able to use a flex body in our simulink model?

yuyihsiang, 2020-06-16:
I agree this is probably an issue with using NEMOH. We have built everything and tested the GBM model based on WAMIT coefficients. Can you share your NEMOH files and the WEC-Sim runs? I should be able to modify and update WEC-Sim as needed, mainly how body.dof_end is defined.

RApmmh, 2020-06-17:
OK so if I understand correctly, GBM in WEC-Sim may not work using NEMOH inputs? 
My model is relatively simple: a cantilevered horizontal flexible membrane. The structure supporting the cantilever is left out of the model. The simulink model only includes the flex body, a rotational constraint and the ground reference frame.
My NEMOH code (attached here) does not include GBM. Ideally, I would have wanted to use NEMOH inputs with one DOF (rotational on the cantilever axis), along with the generalized body modes. However, BEMIO only worked with NEMOH inputs containing exactly 6 DOF, so this is what I used for my WEC-Sim runs (also attached). 

So potentially, if I run my NEMOH code with 1 DOF and the GBMs, then manually write an H5 file, then this could accurately simulate my model with WEC-Sim? 

P.S. I left out the results folder from my NEMOH data because of the attachment size limit
[test4.zip](https://github.com/WEC-Sim/WEC-Sim/files/4791491/test4.zip)

yuyihsiang, 2020-06-17:
Yes, the BEMIO needs some modifications to read in the GBM parts correctly. If you can provide me a NEMOH run with 6 rigid DOF and GBMs, I should be able to modify the BEMIO quickly. We just don't have the NEMOH example for us to do that in the past.

yuyihsiang, 2020-07-22:
If you like, please provide us a NEMOH run with 6 rigid DOF and GBMs results from Nemoh.

RApmmh, 2020-07-23:
Thank you so much for your help. Due to conflicting time constraints I haven't been able to go back to this WEC simulation. I will close the thread for now and re-open when I have the NEMOH files (which also need a special modification in the NEMOH codes to calculate GBMs)

closed 2020-07-23 by RApmmh