WEC-Sim/WEC-Sim #367: Error while running tutorial

kjbennichan, 2020-06-12:
While running tutorials, its coming as initializing the simulation class and then nothing.
WEC-Sim: An open-source code for simulating wave energy converters
Version: NOT DEFINED

Initializing the Simulation Class...
	Case Dir: C:\Program Files\MATLAB\R2016a\bin 
Like this, how to get it run.

Amala

dforbush2, 2020-06-12:
Hello,

The tutorials found in  the `./Tutorials` directory does not run without user modification. The idea is to provide a  template `wecSimInputFile.m` and simulink model file which a new user can populate to familiarize themselves with the WEC-Sim workflow. See http://wec-sim.github.io/WEC-Sim/tutorials.html for tutorial instructions.

 The `./Examples` directory houses working examples that should run without any user modification. If you are attempting to test your installation, please run an example case, as per step 3 http://wec-sim.github.io/WEC-Sim/getting_started.html

Best,
Dom

dforbush2, 2020-06-17:
Did this solve your issue?

closed 2020-06-18 by dforbush2