WEC-Sim/WEC-Sim #370: case directory

kjbennichan, 2020-06-15:
Hi, 
A WEC-Sim input file is required for each run. It MUST be placed inside the case directory
for the run and MUST be named wecSimInputFile.m.
What do you mean by placing inside case directory. How to do that?

dforbush2, 2020-06-15:
Please see the WEC-Sim "Getting Started" step 3 for a simple example.

http://wec-sim.github.io/WEC-Sim/getting_started.html

The "case directory" is the directory you are in when you type `wecSim` into the matlab command window. The case directory must contain a `wecSimInputFile.m` and a simulink model `YOUR_MODEL.slx`

In the example in the link, the case directory would be `$WECSIM/examples/RM3` where $WECSIM is the root directory of your WEC-Sim installation.

dforbush2, 2020-06-17:
Did this solve your issue?

kjbennichan, 2020-06-18:
No I couldn't

dforbush2, 2020-06-18:
Can you be more specific about the problem you are encountering?

kjbennichan, 2020-06-18:
WEC-Sim: An open-source code for simulating wave energy converters
Version: NOT DEFINED

Initializing the Simulation Class...
Case Dir: C:\Users\HP\Documents\WEC-Sim-4.0\tutorials\RM3

Iam not able to run the tutorial RM3, I prepared slx file also, still it's
showing this and then nothing.

dforbush2, 2020-06-18:
Are you able to run the working example case in $WEC-Sim\examples\RM3?

kjbennichan, 2020-06-19:
I can run the bemio.m file, but the wecsiminputfile.m is not running for
the example RM3.
when I am trying wecsiminputfile.m, I am encountering the error I told in
the above mail. Don't know how to rectify that, and make that run.

dforbush2, 2020-06-22:
The message you are encountering is not an error message but a printed output from the `wecSimInputFile.m `. If you are not seeing an additional error message it seems like your run is terminating after running `wecSimInputFile.m`. 

A WEC-Sim simulation is executed by typing `wecSim` into the command window. 

This will first run `wecSimInputFile.m`, and then run the rest of the simulation. Please see https://wec-sim.github.io/WEC-Sim/overview.html#running-wec-sim.

kmruehl, 2020-07-02:
@kjbennichan did this resolve your issue?

closed 2020-07-08 by kmruehl