WEC-Sim/WEC-Sim #372: Errors with Matlab 2019B and 2020A

Aquaharmonics, 2020-06-17:
Hello,

I was updating Matlab to the latest version 2020a and testing with a pull of the latest WEC-Sim version from the git hub repository. 

The same installation runs fine in Matlab 2018b the following version:

```
MATLAB Version: 9.5.0.1298439 (R2018b) Update 7
MATLAB License Number: 40626616
Operating System: Microsoft Windows 10 Pro Version 10.0 (Build 18363)
Java Version: Java 1.8.0_152-b16 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.5         (R2018b)
Simulink                                              Version 9.2         (R2018b)
Control System Toolbox                                Version 10.5        (R2018b)
Optimization Toolbox                                  Version 8.2         (R2018b)
Simscape                                              Version 4.5         (R2018b)
Simscape Multibody                                    Version 6.0         (R2018b)
```

In 2019b, the following Version:
```

> MATLAB Version: 9.7.0.1319299 (R2019b) Update 5
> MATLAB License Number: 40626616
> Operating System: Microsoft Windows 10 Pro Version 10.0 (Build 18363)
> Java Version: Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
> -----------------------------------------------------------------------------------------------------
> MATLAB                                                Version 9.7         (R2019b)
> Simulink                                              Version 10.0        (R2019b)
> Control System Toolbox                                Version 10.7        (R2019b)
> Optimization Toolbox                                  Version 8.4         (R2019b)
> Simscape                                              Version 4.7         (R2019b)
> Simscape Multibody                                    Version 7.0         (R2019b)

```
This error is thrown:

```
Error using wecSim (line 291)
Error due to multiple causes.
Caused by:
    Error using wecSim (line 291)
    Variant control 'sv_b_hydroBody' of variant block 'RM3/Float' must be a condition expression or name of a Simulink.Variant object in the global
    scope workspace of the model. Ensure that the condition does not match the model name.
    Error using wecSim (line 291)
    Variant control 'sv_b_nonHydroBody' of variant block 'RM3/Float' must be a condition expression or name of a Simulink.Variant object in the
    global scope workspace of the model. Ensure that the condition does not match the model name.

```

In 2020a the following version:

```
-----------------------------------------------------------------------------------------------------
MATLAB Version: 9.8.0.1380330 (R2020a) Update 2
MATLAB License Number: 40626616
Operating System: Microsoft Windows 10 Pro Version 10.0 (Build 18363)
Java Version: Java 1.8.0_202-b08 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.8         (R2020a)
Simulink                                              Version 10.1        (R2020a)
Control System Toolbox                                Version 10.8        (R2020a)
Optimization Toolbox                                  Version 8.5         (R2020a)
Simscape                                              Version 4.8         (R2020a)
Simscape Multibody                                    Version 7.1         (R2020a)

```
This error is thrown:

```
Error using wecSim (line 291)
Error due to multiple causes.
Caused by:
    Error using wecSim (line 291)
    Error in 'RM3/Float/Hydrodynamic Body': Failed to evaluate mask initialization commands.
        Error using wecSim (line 291)
        File Solid block (mask) does not have a parameter named 'ExtGeomFileType'
    Error using wecSim (line 291)
    Error in 'RM3/Spar//Plate/Hydrodynamic Body': Failed to evaluate mask initialization commands.
        Error using wecSim (line 291)
        File Solid block (mask) does not have a parameter named 'ExtGeomFileType'
```

When looking at the mask in the mask editor under the initialization tab, I can see where this part in the initilization script is:

```
if exist('body','var')
    % geometry file name
    h=[gcb '/Structure/Body Properties'];
    geomFile=body.geometryFile;
    set_param(h,'ExtGeomFileName',geomFile);
```

[image]

Just wondering if there is something specific to 2019b and 2020a  I need to change or setup for initialization that is different for 2108b and prior that needs to be changed for newer Matlab versions (though I see in the documentation that it has been tested through 2019b) , or if you have seen this kind of error for other reasons.

Also to note, I re-ran BEMIO for the RM3.h5 files just in case they did not download correctly, but seems to be okay as it runs fine in 2018b.

Thanks!

Alex

yuyihsiang, 2020-06-17:
Yes, it is a version issue. Matlab removed the 'ExtGeomFileType' parameter starting 2019b. We have fixed the issue it by adding an additional if statement to handle different version. Please pull the latest version.

kmruehl, 2020-06-17:
@Aquaharmonics Thank you very much for bringing this to our attention. We are aware of this WEC-Sim library issue. You are correct about the route cause, and that it is an issue related to the MATLAB version. 2018b works fine, but 2019b and newer are problematic. What happened is that MATLAB removed the option to select the geometry type (`*.STL` or `*.STEP`). We are actively working on resolving this issue, and are optimistic that the latest commit 1c30048 from @yuyihsiang will resolve the issue.

Aquaharmonics, 2020-06-17:
Thanks for the quick response on this, we will do a pull of the latest version and try this again. 
Best Regards,
Alex

closed 2020-06-18 by kmruehl

Aquaharmonics, 2020-06-18:
Hello,

I just finished checking in 2019b and 2020a, and everything works for the RM3 case!

Thanks again for the quick response and help with this.

Best,
Alex

kmruehl, 2020-06-18:
That's great. Glad @yuyihsiang's commit resolved this library issue.