WEC-Sim/WEC-Sim #374: hydraulic PTO

kjbennichan, 2020-06-18:
hi,
 Do hydraulic PTO option works in matlab 2016a. Is it available.

dforbush2, 2020-06-18:
Hi @kjbennichan 

As mentioned in #368, WEC-Sim should support all versions back to 2016a.

Is there a specific error you are encountering?

Best,
Dom

kjbennichan, 2020-06-18:
Hi,
 I don't know how to provide a hydraulic PTO. These are the options
obtained, it's in the attachment. How can I give hydraulic PTO.

dforbush2, 2020-06-18:
Please see applications case: WEC-Sim_Applications\PTO-Sim\RM3_Hydraulic_PTO

The hydraulic PTO parameters are specified in the ``ptoSimInputFile.m``, and define the parameters in the "PTO-Sim" block of the accompanying ``RM3_Hydraulic_PTO.slx`` model.

kmruehl, 2020-07-02:
@kjbennichan did this resolve your issue?

closed 2020-07-08 by kmruehl