WEC-Sim/WEC-Sim #375: OrcaFlex BEMIO inport

bradling, 2020-06-18:
Now that the latest version of OrcaFlex (11.0) includes a BEM solver (OrcaWave), are there any plans to add a function to BEMIO to read the OrcaWave BEM data?

kmruehl, 2020-07-02:
@bradling thanks for bringing this to our attention. Adding `READ_OrcaWave` functionality to BEMIO is not something we currently have on our to do list. However, this is not the first request we've received to do add this feature, so we will consider adding it next FY. In the meantime, you are always welcome to submit a PR to add this functionality :). 

Do you have some example OrcaWave output files you would be able to share with the WEC-Sim team?

bradling, 2020-07-02:
I can share a two-body output file privately with the WEC-Sim team. I will
email you offline about sharing the file.

I have started working on a function that meets my needs. One of the
difficulties is going to be that OrcaWave calculates hydrodynamic
coefficients at [0,0,0], instead of at the body CG. Doing the kinematic
transformation correctly for any arbitrary body location will take some
time to get correct. If what I end up writing to meet my needs is
generalizable enough, I will plan on making a pull request containing the
function.

kmruehl, 2020-07-08:
@bradling thank you very much for providing sample OrcaWave files. We will consider adding this feature. 

I've noted your point about generalizing the requested bemio functionality to convert from the output coordinate system at [0,0,0] to the cg.

akeeste, 2021-09-15:
As an alternative --
We could add a body parameter that defines the relative position between the cg and where hydrodata is defined. Then we can add a rigid transform in Simulink to apply the hydro forces where the BEM defines them, as opposed to creating a complex kinematic transform.

closed 2021-11-03 by dav-og