WEC-Sim/WEC-Sim #376: error

kjbennichan, 2020-06-19:
hi,
WEC-Sim Read Input File ...   
Undefined function or variable 'value'.

Error in simulationClass/checkinputs (line 135)
                error('The simMecahnics file, %s, does not exist in the case
                directory',value)

Error in wecSim (line 43)
simu.checkinputs;
 This is the error I am encountering while running OSWEC tutorial, how can i rectify this?
>>

dforbush2, 2020-06-22:
This error indicates that the Simulink `.slx` file specified in `wecSimInputFile.m` does not exist in the case directory.

Please note that tutorials are NOT working examples and are intended to familiarize the user with the WEC-Sim workflow, and must be modified by the user before they will be functional. 
By default the OSWEC tutorial case directory contains the empty model `OSWEC.slx`, but this name is not specified on line 3 of the `wecSimInputFile.m `.

Please see https://wec-sim.github.io/WEC-Sim/tutorials.html#oscillating-surge-wec-oswec for step-by-step instructions on how to set up and run this tutorial case.

kmruehl, 2020-06-24:
@kjbennichan if you're looking for a working example of the OSWEC, you can run the `$SWECSIM/examples/OSWEC` case located here: 
https://github.com/WEC-Sim/WEC-Sim/tree/master/examples/OSWEC 

As @dforbush2 mentioned, the tutorials are not working examples. They are meant to show users how to set up and run a WEC-Sim model.

closed 2020-06-24 by kmruehl