WEC-Sim/WEC-Sim #377: for the default RM3 point absorber

abaddonwang, 2020-06-23:
I run the RM3 point absorber example successfully, the default run time is 400s with 2.5m wave height and 8s wave period regular wave (CIC), the result is good and stable, however, when I increased the run time to a longer time, for example 4000s, then the results starts to lose stability from about 1000s, it had a sudden change and the device starts to move wildly.

I kept all the default settings except from the run time. So does anybody know how to solve the unstable problem?

yuyihsiang, 2020-07-09:
This is most likely an issue because of how the added mass term was included in the solver. There are two ways of solving the issue.

- The easier way to do is to change the simu.adjMassWeightFun value to a larger number. The default value is 2, and change it to  simu.adjMassWeightFun=10 will solve the issue.
- Assign viscous damping to the body. The viscous effect was neglected in the example.  

--Yi-Hsiang

abaddonwang, 2020-07-10:
Good morning Yi-Hsiang,

Thank you very much, I will try.

Best regards,

Daming

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

This email and any files with it are confidential and intended solely for the use of the recipient to whom it is addressed. If you are not the intended recipient then copying, distribution or other use of the information contained is strictly prohibited and you should not rely on it. If you have received this email in error please let the sender know immediately and delete it from your system(s). Internet emails are not necessarily secure. While we take every care, University of Plymouth accepts no responsibility for viruses and it is your responsibility to scan emails and their attachments. University of Plymouth does not accept responsibility for any changes made after it was sent. Nothing in this email or its attachments constitutes an order for goods or services unless accompanied by an official order form.

abaddonwang, 2020-07-10:
Hi, Yi-Hsiang,

I have another question, when I turned on the non-linear analysis by setting simu.nlHydro = 2; the result is very unstable, the responses is not as periodic as linear analysis,. And from the animation, it seems the whole model floated away.  I don't know why the non-linear result is so different from the linear cases

 I tried to constrain the RM3 model, the example allows the spar to move in 3DOF (surge, heave,pitch). I found that if I restricted the spar in surge (only free to move in heave and pitch for the spar), the result is stable again. I also tried the model with spar completed fixed, so the float can only move vertically, the results is also stable.

it seems like that if we want to have a reasonable non-linear result, we have to introduce some mooring system to constrain it ,is that right? My supervisor wants me to analyze the non-linear cases, but the unstable results confused me.

Best regards,

Daming

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

kmruehl, 2020-07-13:
@abaddonwang this is due to the fact that non-linear hydro uses the `*.stl` geometry files to determine the volumetric displacement and buoyancy.  This means that good `*.stl` mesh resolution is required for the WEC geometry files when using the nonlinear buoyancy and Froude-Krylov wave excitation in WEC-Sim. The RM3 `*.stl` files provided in the example were developed only for visualization only, and do not have the required mesh resolution to run with `simu.nlHydro = 2`. However, we do provide an [example of running non-linear hydro for an ellipsoid on the applications repository](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Nonlinear_Hydro). 

For more information about `simu.nlHydro` and its requiremetns, refer to the [non-linear hydro online documentation](http://wec-sim.github.io/WEC-Sim/advanced_features.html#nonlinear-buoyancy-and-froude-krylov-excitation).
.

yuyihsiang, 2020-07-22:
I just want to add that, yes, you will need a mooring to keep the device in position. The nonlinear hydro effect will result in a drift force.

yuyihsiang, 2020-07-22:
did this resolve your issue?

abaddonwang, 2020-07-23:
Thank you, I think you solved my problem.

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

closed 2020-08-05 by kmruehl