WEC-Sim/WEC-Sim #380: Paraview Question

Aquaharmonics, 2020-07-02:
Hi,

We were trying out Paraview visualization which we have not done for awhile, and had a **question.**

Does the  writing of the Paraview files typically take a long time? There does not seem to be any other errors we can see, and files are being written to the directory but it seems to be taking awhile (relative to the WecSim solution time), say over an hour to write the files using a 0.2 second output time step for a 240 second WecSim simulation. We have not run anything in Paraview yet, but wanted to check if this is normal or not, or if there are things we should change to improve this.

Best,

Alex

kmruehl, 2020-07-02:
@Aquaharmonics unfortunately, yes writing the Paraview files at each time step does slow down the simulation substantially. Fortunately, there are some new `simulationClass` options submitted in PR #355 and merged in f55c5fb93a71391bd5762111ccc6bd51da96b6e7 that should speed up the `write_paraview` methods. You can now specify the time-steps you would like to write ParaView VTP files for, and specify start and end times:

http://wec-sim.github.io/WEC-Sim/api.html#source.objects.simulationClass.dtParaview
http://wec-sim.github.io/WEC-Sim/api.html#source.objects.simulationClass.StartTimeParaview
http://wec-sim.github.io/WEC-Sim/api.html#source.objects.simulationClass.EndTimeParaview

Aquaharmonics, 2020-07-02:
@kmruehl,

Thanks for the quick response, it is helpful we just wanted to make sure there wasn't something else going on. 

Best,
Alex

kmruehl, 2020-07-03:
@Aquaharmonics nothing else going on, writing those files is just slow.

closed 2020-07-03 by kmruehl