WEC-Sim/WEC-Sim #381: adding two PTO-sim to one simullink proyect

ignacio-byte, 2020-07-03:
is it posible to add a secong power take of in one system... two hidraulic cilinders in one device for example adding a two hidraulic cilinders to the oswec,

enrjcutl, 2020-07-28:
Is there any update to this? I am trying to simulate two WECs attached to one floating platform and wondering whether it is all possible.

dforbush2, 2020-07-28:
Hi all,

Sorry for the delay. This should be possible within WEC-Sim using either the ready-made PTO blocks, PTO-Sim, or (if the toolboxes are available), elsewhere in Simscape. Are you encountering a specific error when you try to do this? To give more specific help I would need to see your models, but generally, just make sure that the PTO parameters are defined for each in your wecSim input file. As an example using the ready-made PTO blocks, your input file should contain:

`% first PTO
pto(1) = ptoClass('PTO1')
pto(1).k = 0; % whatever value you want
pto(1).c = 120000; % whatever value you want
pto(1).loc = [0 0 0]; % whatever location you want
% second PTO
pto(2) = ptoClass('PTO2')
pto(2).k = 0; % whatever value you want
pto(2).c = 120000; % whatever value you want
pto(2).loc = [0 0 0]; % whatever location you want`

and your *.slx model should contain two pto blocks named 'pto1' and 'pto2' respectively.

dforbush2, 2020-07-28:
With particular respect to the OSWEC question, I would point you to the WEC-Sim_Applications/PTO-Sim/OSWEC/OSWEC_Hydraulic_PTO example as a starting point. You will need to modify the PTO-sim input file and the simulink file such that each PTO component block has a unique identifier number across each of your PTOs and the appropriate parameters are specified for each. For example. if you were to use two identical PTOs like in the above example, you would end up with a ptoSimInputFile like the attached.
[ptoSimInputFile.txt](https://github.com/WEC-Sim/WEC-Sim/files/4990554/ptoSimInputFile.txt)

I did run into an error in the PTO-sim library on the "valve" block that would cause problems with multiple instances. I am pushing a commit that will update the PTO-Sim library to fix this so please pull a fresh clone.

dforbush2, 2020-07-30:
Are there any follow-up questions? Did this solve your issue?

ignacio-byte, 2020-07-30:
> Are there any follow-up questions? Did this solve your issue?

yes,i did exactly what you wrote before and it works well for me,but i have a problem trying to add linearDamping 

Error using wecSim (line 261)
Error due to multiple causes.
Caused by:
    Error using wecSim (line 261)
    Matrix multiply dimensions propagation error. Error occurred while setting input port 2 of
    'dcc_cHydraulic_PTO2/alaizquierda/Hydrodynamic Body/Additional Linear Damping Force Calculation/Product1' to have
    the dimensions [6]. A possible cause for this error is that these dimensions do not agree with the partial
    dimensions information present on other ports
    Error using wecSim (line 261)
    Error in port widths or dimensions. Output port 1 of 'dcc_cHydraulic_PTO2/alaizquierda/Hydrodynamic Body/Additional
    Linear Damping Force Calculation/Velocity' is a one dimensional vector with 6 elements.

i delete linear damping and the model runs,but for a good simulation i need the linear damping, is it possible the error may caused by the version of matlab(2019a) or ansys aqwa (18.1)

dforbush2, 2020-07-30:
I believe that the issue could be due to a recent change to the WEC-sim library that became necessary due to changes in Simulink defaults in a recent MATLAB edition. It seems that this change has not been properly propagated throughout our example WEC-Sim Applications input files, so thank you for alerting us to this. I'll make sure to address this in short order.

In the mean-time, changing the definition of the linear damper from a 6 element vector to a  6 x 6 diagonal matrix should solve the problem. For example change the input file line FROM

`body(1).linearDamping = [0 0 0 0 1E7 0];`

TO

`body(1).linearDamping = diag([0 0 0 0 1E7 0]);`

ignacio-byte, 2020-07-30:
> I believe that the issue could be due to a recent change to the WEC-sim library that became necessary due to changes in Simulink defaults in a recent MATLAB edition. It seems that this change has [...]

that new way worked! i have a doutb with that value,i know it belongs to the oswec in your tutorials,how did you get 10.000.000   in Rx? i mean what's your formula?

dforbush2, 2020-07-30:
I'm not sure to be honest, I just borrowed the one from the tutorial. In simulations without a control law to apply a force resisting the excitation, the linear or quadratic damping is selected to be large enough to counteract the wave excitation force at while the device undergoes reasonable displacement.

ignacio-byte, 2020-07-30:
> I'm not sure to be honest, I just borrowed the one from the tutorial. In simulations without a control law to apply a force resisting the excitation, the linear or quadratic damping is selected to [...]

so i can pick a value in linear damping that let me get a nice displacement for example,to protect hidraulic cilinders of the device?

dforbush2, 2020-07-30:
Since you seem to be modeling a power take-off, which should act in many circumstances like a damper, I would recommend that you review the PTO-sim blocks for the components you are using and prescribe the damping appropriate for the drivetrain you are trying to model, since that is where hopefully most of the damping in a system will originate from: the sum of a prescribed actuation force and losses in each component. 

For example, the OSWEC_Hydraulic_PTO model we've been discussing, the Rotary Generator Lookup Table block prescribes generator torque based on a generator speed: this acts effectively like a damping torque. 

I would recommend that you use additional damping terms like body(1).linearDamping and/or body(1).viscDrag.cd and body(1).viscDrag.characteristicArea to capture damping/drag effects external to the PTO, such as those exerted by the fluid on the body. This will most closely approximate reality and make your PTO/controller results more meaningful.

ignacio-byte, 2020-07-30:
yes,I was trying linear damping, PTO damping and viscous drag,but its very
difficult to determinate viscous drag for the geometry of mi model,
its  3m high x 12 m long x 2 m width  (each wing) its like a rhombus if u
could give me a help :D the model is something inusual

%% Simulation Data
simu = simulationClass();               % Initialize Simulation Class
simu.simMechanicsFile = 'dcc_cHydraulic_PTO2';    % Specify Simulink Model File
simu.mode = 'normal';                   % Specify Simulation Mode ('normal','accelerator','rapid-accelerator')
simu.explorer='on';                     % Turn SimMechanics Explorer (on/off)
simu.startTime = 0;                     % Simulation Start Time [s]
simu.rampTime = 100;                    % Wave Ramp Time [s]
simu.endTime = 400;                     % Simulation End Time [s]        
simu.solver = 'ode4';                   % simu.solver = 'ode4' for fixed step & simu.solver = 'ode45' for variable step 
simu.dt = 0.01;                         % Simulation Time-Step [s]
simu.CITime = 30;                       % Specify CI Time [s]
%simu.pressureDis = 1;                   % Saves pressures data for Paraview
simu.b2b = 1;
%simu.paraview = 1;                      % Saves data to *.vtp for Paraview
%% Wave Information
% % noWaveCIC, no waves with radiation CIC  
% waves = waveClass('noWaveCIC');       % Initialize Wave Class and Specify Type  

% % Regular Waves 
waves = waveClass('regular');           % Initialize Wave Class and Specify Type                                 
waves.H = 1.5;                          % Wave Height [m]
waves.T = 9;                            % Wave Period [s]

% Irregular Waves using PM Spectrum with Directionality 
%waves = waveClass('irregular');         % Initialize Wave Class and Specify Type
%waves.H = 5;                          % Significant Wave Height [m]
%waves.T = 12;                            % Peak Period [s]
%waves.spectrumType = 'PM';              % Specify Spectrum Type
%waves.waveDir = [0,30,90];              % Wave Directionality [deg]
%waves.waveSpread = [0.1,0.2,0.7];       % Wave Directional Spreading [%}

% Irregular Waves with imported spectrum
%waves = waveClass('spectrumImport');        % Create the Wave Variable and Specify Type
%waves.spectrumDataFile = 'spectrumData.mat';  %Name of User-Defined Spectrum File [:,2] = [f, Sf]

% % Waves with imported wave elevation time-history  
%waves = waveClass('etaImport');         % Create the Wave Variable and Specify Type
%waves.etaDataFile = 'etaData.mat'; % Name of User-Defined Time-Series File [:,2] = [time, eta]

%% Body Data
%alaizquierda
body(1) = bodyClass('hydroData/ANALYSIS.h5');        % Initialize bodyClass for Flap
body(1).geometryFile = 'geometry/IZQUIERDA.stl';     % Geometry File
body(1).mass = 4301.3;                               % User-Defined mass [kg]
body(1).momOfInertia = [56898.79 6342.22 53889.46];  % Moment of Inertia [kg-m^2]
body(1).linearDamping = diag([0 0 0 0 0 1E6]);       % Specify damping on body      % Linear Viscous Drag Coefficient

body(1).viscDrag.cd = [1 0 0 0 0 1];
body(1).viscDrag.characteristicArea = [36 0 0 0 0 36];

% aladerecha
body(2) = bodyClass('hydroData/ANALYSIS.h5');        % Initialize bodyClass for Flap
body(2).geometryFile = 'geometry/DERECHA.stl';       % Geometry File
body(2).mass = 4301.3;                               % User-Defined mass [kg]
body(2).momOfInertia = [56898.79 6342.22 53889.46];  % Moment of Inertia [kg-m^2]
body(1).linearDamping = diag([0 0 0 0 0 1E6]);       % Specify damping on body      % Linear Viscous Drag Coefficient
%body(2).viscDrag.cd 
%body(2).viscDrag.characteristicArea 
%base

body(3) = bodyClass('hydroData/ANALYSIS.h5');      % Initialize bodyClass for Flap
body(3).geometryFile = 'geometry/base.stl';     % Geometry File
body(3).mass = 'fixed';    

%marco
[... 35 lines cut ...]
%Rotational PTO3 base-marco
pto(3) = ptoClass('PTO3');                      % Initialize ptoClass for PTO1
pto(3).k = 50000000;                            % PTO Stiffness Coeff [Nm/rad]
pto(3).c = 5000000;                             % PTO Damping Coeff [Nsm/rad]
pto(3).loc = [12 0 -2.5];                       % PTO Location [m] [0 0.51 -2.5]; 

%% Mooring
% Moordyn
%mooring(1) = mooringClass('mooring');       % Initialize mooringClass
%mooring(1).moorDynLines = 6;                % Specify number of lines
%mooring(1).moorDynNodes(1:3) = 16;          % Specify number of nodes per line
%mooring(1).moorDynNodes(4:6) = 6;           % Specify number of nodes per line
%mooring(1).initDisp.initLinDisp = [0 0 -0.21];      % Initial Displacement
% Rotational base-marco
%pto(3) = ptoClass('PTO3');                      % Initialize ptoClass for PTO1
%pto(3).k = 100000000;                                   % PTO Stiffness Coeff [Nm/rad]
%pto(3).c = 10000000;                               % PTO Damping Coeff [Nsm/rad]
%pto(3).loc = [11.6 0 -4.99];                       % PTO Location [m] [0 0.51 -2.5]; 

% Mooring Matrix
%mooring(1) = mooringClass('mooring');       % Initialize mooringClass
%mooring(1).matrix.k = zeros(6,6);
%mooring(1).matrix.k(1,1) = 1e5;
%mooring(1).matrix.c = zeros(6,6);
%mooring(1).matrix.preTension = zeros(1,6);

dforbush2, 2020-08-05:
Did this solve your issue?

closed 2020-08-18 by dforbush2