WEC-Sim/WEC-Sim #385: Point absorber WEC

adzanaihs, 2020-07-08:
I'm new to this program and currently trying to use it to simulate WEC similar to the design used by Wavestar Energy. But instead of using hydraulic PTO, I'm using direct mechanical drive PTO. Is that possible to simulate using this program? Because from the tutorial on the website there are only two examples of WEC (two-body point absorber and OSWEC).

[image]

kmruehl, 2020-07-08:
@adzanaihs yes, it is possible to simulate a device like the Wavestar using WEC-Sim. We actually have an open access model of a Wavetsar-like device that was used for the [WEC Control Competition (WECCCOMP)](https://github.com/WEC-Sim/WECCCOMP). I recommend starting with the WECCCOMP model as your baseline and modifying it to meet your needs. FYI, we also have a repository of[ WEC-Sim applications](https://github.com/WEC-Sim/WEC-Sim_Applications) in addition to the RM3 and OSWEC examples.

kmruehl, 2020-07-22:
@adzanaihs did this resolve your issue?

closed 2020-08-05 by kmruehl