WEC-Sim/WEC-Sim #387: How to read hydro.B variable?

t-osu, 2020-07-20:
Evening,

I'm using WEC-Sim to verify my own IRF calculations from AQWA data, a 2 body point absorber. So far my excitation IRF matches the output from WEC-Sim, but I am confused on how to read the hydro.B variable for the radiation IRF. It is a 12x12x100. I understand the 3rd dimension, it's the first two that are giving me difficulties. 

Is it the first 6 rows correspond to the first body, broken down into rows 1 through 6 as heave, sway, surge, roll, pitch, yaw? The latter 6 rows following the same pattern, but for the second body?
I think this is what I assumed and I was unable to produce the same radiation IRF WEC-Sim calculates.

Any guidance is appreciated.

- Chris

kmruehl, 2020-07-22:
@t-osu the top left [6,6] matrix of `hydro.B` is the frequency domain radiation damping corresponding to surge, sway, heave, roll, pitch and yaw for Body 1, and the bottom right [6,6] matrix of `hydro.B` corresponds to surge, sway, heave, roll, pitch and yaw for Body 2. The off-diagonal [6,6] matrices correspond to the cross-coupling terms between Body 1 and Body 2.

kmruehl, 2020-08-05:
@t-osu Does this resolve your issue?

t-osu, 2020-08-06:
Yes, I was actually able to confirm this on my own before the reply, but glad to know I interpreted it the right way!

kmruehl, 2020-08-06:
@t-osu glad to hear that. I'm going to go ahead and close this issue then.

closed 2020-08-18 by kmruehl