WEC-Sim/WEC-Sim #388: How to use MoorDyn?

ZhiGN, 2020-07-20:
I am a novice programmer. I can run the MoorDyn program, but the FAIRTEN in the result file Lines.out is deviated from the FairTens result generated by the matlab calculation example in the workspace.
[image]

kmruehl, 2020-07-22:
@ZhiGN can you upload this WEC-Sim/MoorDyn model so that we can debug this issue?

yuyihsiang, 2020-07-22:
Can you upload the files for us to check?

--Yi-Hsiang

ZhiGN, 2020-07-25:
@kmruehl @yuyihsiang First of all thank you for your kind reply. This file is an example of moordyn published by Professor Hall. I haven't combined moordyn and wec-sim. When running separately, I found that the FAIRTEN in the result file Lines.out is different from the value output by the scope module in simulink, but I feel that the two values should be the same. I have seen the source program, but because I don't understand C++ programs, I didn't find the reason. Thank you again for your kind reply.
[Simulink.zip](https://github.com/WEC-Sim/WEC-Sim/files/4974853/Simulink.zip)
[MoorDyn v1.00.03C Source.zip](https://github.com/WEC-Sim/WEC-Sim/files/4974854/MoorDyn.v1.00.03C.Source.zip)

yuyihsiang, 2020-07-28:
Thanks for uploading the data. I have asked Matt to take a look at this.

mattEhall, 2020-09-02:
Hi @ZhiGN, there was an error in earlier versions of MoorDyn (half the weight of the top mooring line segment was neglected in the fairlead force calculation) that might explain the disagreement you're seeing. It was fixed a few years ago in version 1.01.01C. It looks like you're using an earlier version, so my recommendation would be to try v1.01.01C and see if it solves the problem.

kmruehl, 2020-09-21:
I've added this issue to upgrading WEC-Sim to MoorDyn2 on the project board: https://github.com/WEC-Sim/WEC-Sim/projects/28

yuyihsiang, 2021-08-04:
WEC-Sim/MoorDyn has been updated to the latest release version.

--Yi-Hsiang

closed 2021-08-04 by yuyihsiang

HANloading, 2024-02-22:
> I am a novice programmer. I can run the MoorDyn program, but the FAIRTEN in the result file Lines.out is deviated from the FairTens result generated by the matlab calculation example in the workspace. [image]

你好，请问moordyn的调用解决了吗？？，能否加个联系方式？