WEC-Sim/WEC-Sim #391: Visualisation and simulation problems

enrjcutl, 2020-07-28:
Hello,

I am new to WEC-Sim. I have created my own version of the RM3 tutorial. I developed the geometry in Solidworks and obtained the hydrodynamic properties via ANSYS AQWA. I have used the same .slx and input file as the tutorial and the results are the same. There are two issues however, the first is that my geometry does not line up with the environment. I have seen previous posts regarding this issue but I cannot figure this one out. Secondly, during the simulation the float remains fixed to the spar and it is only the spar that moves in the 3 DOF as instructed. Could someone explain how to resolve this please? Many thanks in advance.
[image]

dforbush2, 2020-07-28:
Hi @enrjcutl 

I believe position of the body relative to the origin in the visualization is based upon the [0,0,0] origin specified in the solidworks model. Unless I'm being deliberately diligent about sketching my geometry in solidworks, its really easy to define your sketch (and thus, geometry) at some point far from this origin, and the resulting WEC-Sim visualization can appear similar to the screenshot you have provided. If this is indeed the problem, I would recommend either re-centering your solidworks model about the appropriate origin, or using a tool such as meshmagick https://lheea.github.io/meshmagick/command_line/cli_usage.html#translations to translate the mesh appropriately. 

To your second point, the default RM3 example has a  3 DOF (heave, surge, pitch) floating constraint between the spar and the seabed, but a single DOF (heave) linear power-take off constraining the motion between the float and the spar to translation along the spar's long axis. If your intent was to replicate the default RM3 example, the behavior you are describing is expected.

enrjcutl, 2020-07-28:
Hi dforbush2,

This is where it becomes confusing because I was extremely diligent in aligning with the origin. In AQWA, the alignment was perfect however in WEC-Sim not so, I will have another try and let you know the outcome. I believe it is something to do with the .stl files. 

I understand that, however the float itself does not translate along the spar at all, it remains fixed for all wave conditions. 
Thank you for your response.

enrjcutl, 2020-07-28:
This is the origin on Solidworks.
[image]

dforbush2, 2020-07-28:
Can you please provide the model and necessary input files? There are a number of possibilities here. But a quick check would be to make sure the PTO damping (pto.c) and stiffness (pto.k) are not overly high, and that you are selecting a wave state that is preferentially exciting floats.

I'm not sure about the visualization. I will look into this.

enrjcutl, 2020-07-28:
[Tutorial.zip](https://github.com/WEC-Sim/WEC-Sim/files/4990819/Tutorial.zip)

dav-og, 2020-07-29:
Hello, thanks for providing these files.

I've just checked your .stl files with meshmagick. We can infer dimensions/centres from xmin, xmax etc:

        --------------------------------------------
                MESH NAME : float
        --------------------------------------------

        Number of vertices: 254
        Number of faces:    508

        Number of triangles:   508
        Number of quadrangles: 0

        xmin = 10906.179688     xmax = 31299.398438
        ymin = 10906.179688     ymax = 31299.398438
        zmin = 27401.000000     zmax = 32401.000000

        --------------------------------------------
                MESH NAME : plate
        --------------------------------------------

        Number of vertices: 180
        Number of faces:    356

        Number of triangles:   356
        Number of quadrangles: 0

        xmin = 6103.211426      xmax = 36102.367188
        ymin = 6103.211426      ymax = 36102.367188
        zmin = 300.998688       zmax = 38401.000000

- It seems like the .stl files have dimensions in mm? They need to be in m.
- The meshes seem to have experienced positive translations in x, y and z. I can't say how that has happened, only what I can see from the .stl files. It actually seems like the bottom of the plate is defined close to the waterline (although not quite - there is still a small gap):

[image]

- I would recommend trying to identify what has gone wrong in the export from SolidWorks (e.g. 'mm' in settings...not sure about translations)
- Your other option would be to download meshmagick and obtain something closer to what you intended by applying scaling/translation operations

Cheers,
Dave

dav-og, 2020-07-29:
Also, on this point:

`I understand that, however the float itself does not translate along the spar at all, it remains fixed for all wave conditions.`

I checked your output file - WEC-Sim is computing body motion in the order of ~1m - that's just not perceptible in the animations because the .stl geometries are so large (10s of km).

Cheers,
Dave

enrjcutl, 2020-07-30:
Hi Dave,

Thanks for your reply. 

I am really not sure what is causing this. The geometry in Solidworks itself is correct, units, location etc. However as soon as it is exported into an .stl format and that file is reopened this is where the problems arise. I am not sure what is causing this, even the export units are in meters for the .stl format. It is bizarre and I am now looking to troubleshoot this. 

Kind Regards,
Josh

enrjcutl, 2020-07-30:
I have now resolved the issue and have a working model. 

Many Thanks.

dav-og, 2020-07-30:
Excellent!

closed 2020-07-30 by dav-og

sugarl14, 2020-08-10:
Hi @enrjcutl ,

I am experiencing a similar issue with a disproportionate model (the device is much larger than the environment). Could you explain how you solved your issue?

enrjcutl, 2020-08-10:
Hi @sugarl14,

Are you modelling in Solidworks? 
I recall when I was saving the .stl files, the format was wrong and I had to manually change this. When saving the model, click the options button and hopefully a screen will pop up similar to the one below. 
1. Check output units - for myself default was mm even though model was m.
2. Check the "Do not translate STL output data to positive space" - This moves the whole model to all positive coordinates and is the reason why our models are so far from the environment. 

Hope this helps.

[image]

kmruehl, 2020-08-10:
@enrjcutl Thanks for this response!