WEC-Sim/WEC-Sim #395: INSTALLATION PROBLEM

Carbon360, 2020-08-15:
Hi, im using this wecsim for the very first time. I read the manual and ran the code a lot of times. But im getting some error in that each time i run. To start with i just want to install the correct wec sim and use the example versions. Pls help im noob in this. 
Please share if u have developed any tutorial video for installing wec sim and running examples alone.

Carbon360, 2020-08-17:
and for your information im using MATLAB 2019a version

crobarcro, 2020-08-17:
I'm not one of the WEC-Sim developers, just another user, but how do you expect them to help you when you give zero information about the problem? You haven't even said what error message you're getting!

Carbon360, 2020-08-17:
Dear,
My problem is how to proceed the first step. That is the startup folder
point which I didn't understand at all. Pls help me if u can with this.

Carbon360, 2020-08-17:
I can understand that all errors im getting is because i didnt do the step 1 of installation right

dforbush2, 2020-08-17:
As @crobarcro mentioned, it's pretty hard to give you much constructive feedback without having more details. 
Which installation option (clone, fork, or static code download, https://wec-sim.github.io/WEC-Sim/getting_started.html) are you attempting? I would recommend, for debugging, that you try option 3 since there are fewer moving parts. 

Also, can you please provide the error message you are getting? I'm assuming you are referring to a matlab error message that appears after you have entered 'wecSim' in the command window. In which case, copy/pasting that message into this chat or providing a screenshot would be helpful. A detailed description of when the error occurs, what steps (if any) you have tried already to resolve the issue, and any particular points of confusion in the installation process should give us a better place to start.

Carbon360, 2020-08-18:
Dear @dforbush2 im happy that you have heard my request.
So to start with pls forget about the error message which i got earlier as of now.
As per your suggestion i have downloaded the static code of wecsim now. But, now i cannot understand the first step in installation that is... (Rename the wecSimStartup.m file startup.m and paste it into the MATLAB Startup Folder. Set <wecSim> to the $WECSIM/source directory, save the revised startup.m file, and restart MATLAB. Verify the path was set up correctly by checking that the WEC-Sim source directory, $WECSIM/source, is listed in the MATLAB search path. This is done by typing path in the MATLAB).

I dont understand where is the startup.m?

So i will be happy if u eloborate this step on how to do it. Im using MATLAB 2019a version.

crobarcro, 2020-08-18:
You can find information on startup.m here:

https://www.mathworks.com/help/matlab/ref/startup.html

Here's how I found this info:

https://lmgtfy.com/?q=matlab+startup.m

Carbon360, 2020-08-18:
Please tell whether im right or wrong...
as per my understanding, i must create a startup.m file. Then i must copy the code in wecsimstartup.m and paste it in startup.m.
In the copied code, the<wecSim> must be replaced with "$WECSIM/source" and the save it in startup.m and restart matlab.
And also rename the wecsimstartup.m to startup.m

Is my understanding of this right or wrong?

Carbon360, 2020-08-18:
If i did that, i entered 
cd $WECSIM/examples/RM3 
wecSim
and i got error like 
Error using cd
Cannot CD to C:\Users\R HEMANTH PRASANNA\Documents\MATLAB\WEC-Sim-4.1\WEC-Sim-4.1\$WECSIM\examples\RM3
(Name is nonexistent or not a directory).

dforbush2, 2020-08-18:
thanks @crobarcro for your help!

@crobarcro has given the correct advice. It looks as though you are not properly creating your startup.m file. I see two issues with what you have done.

Firstly, the $WECSIM that we use in the instructions is not meant literally as "$WECSIM" but rather as a placeholder for the directory of your WEC-Sim installation. To figure this out, navigate in your matlab window to the top-level of your wec-sim installation, as per the screenshot.

[image]

So for me, my root WEC-Sim directory is C:\Users\dforbus\WEC-Sim, and I would use that wherever "$WECSIM" appears.

From your last post, I am guessing your root WEC-Sim directory is C:\Users\R HEMANTH PRASANNA\Documents\MATLAB\WEC-Sim-4.1\WEC-Sim-4.1\WEC-Sim . This should be used as $WECSIM wherever it appears. 

Secondly, please try placing your startup.m file in your root MATLAB directory. Based on your last post, it appears that this should be C:\Users\R HEMANTH PRASANNA\Documents\MATLAB. 

So your startup.m file should be in your MATLAB root directory and read (at a minimum for WEC-Sim).
```
wecSimPath = '$WECSIM';
addpath(genpath(wecSimPath));

```

Where you have replaced $WECSIM with your WEC-Sim directory string, as discussed above.

Carbon360, 2020-08-18:
Thank so much for your time.
As per your suggestion i did so... i saved startup.m in the C:\Users\R HEMANTH PRASANNA\Documents\MATLAB. see the screenshot.

And i also renamed $WECSIM  with the (C:\Users\R HEMANTH PRASANNA\Documents\MATLAB\WEC-Sim-4.1\WEC-Sim-4.1). Pls see screenshot. 
Im receiving error like "Error using cd
Too many input arguments"
[image]
[image]

.

dforbush2, 2020-08-18:
You have spaces in your directory path name. Generally this should be avoided, but in this particular case you can work around by
`
cd('C:\Users\R HEMANTH PRASANNA\Documents\MATLAB\WEC-Sim-4.1\WEC-Sim-4.1\examples\rm3')`

For the future, this forum is dedicated only to WEC-Sim specific issues. For general matlab errors, https://www.mathworks.com/help/matlab/.

Carbon360, 2020-08-19:
Thanks you helped me resolve all problems. Its fine now.
Now i have 2 questions
1. I need to know another thing which is can i vary the wave height , time period and damping coefficient in the code given.
2. how to get or plot power generated graph in the example model which i simulated.

Thanks.
[image]

dforbush2, 2020-08-19:
continued as issue #401

closed 2020-08-19 by dforbush2