WEC-Sim/WEC-Sim #396: a Question

hossein-mehdipour, 2020-08-15:
I was wondering if there is any specified range for Damping and Stiffness Coefficients in OSWECs, I didn't find anything in the Manual, and wanted to ask you if you could help me with this?

dforbush2, 2020-08-17:
Are you having a particular problem with the example models?

Hydrodynamic damping and stiffness coefficients come from the boundary element method solution, a function of the flap geometry. Additional damping and stiffness parameters for the PTO, for example (from $\examples\OSWEC\wecSimInputFile.m) 

```
% Rotational PTO
pto(1) = ptoClass('PTO1');                      % Initialize ptoClass for PTO1
pto(1).k = 0;                                   % PTO Stiffness Coeff [Nm/rad]
pto(1).c = 12000;                               % PTO Damping Coeff [Nsm/rad]
pto(1).loc = [0 0 -8.9];                        % PTO Location [m]

```
I can't speak exactly to how these parameters were selected, but there is a large body of controls research that can advise selection of these parameters. From a simulation perspective, some > 0 damping is helpful to insure the simulation is stable.

yuyihsiang, 2020-08-24:
In addition to PTO damping, you can specify viscous damping to account for the effect of viscosity on the body motion. There are two ways of doing it. You can specify a rotational viscous damping or use Morison element method. See user manual for more details: https://wec-sim.github.io/WEC-Sim/theory.html#viscous-damping-and-morison-elements

The viscous damping coefficients usually come from empirical solutions or CFD simulations or tunned with experimental data [1]. Two examples of how viscous damping was applied for the OSWEC can be found in [2, 3]. In which one used Morison element method and the other one used a rotational viscous damping coefficient.

[1] van ’t Hoff, J. (2009). Hydrodynamic Modelling of the Oscillating Wave Surge Converter. In The Queen’s University of Belfast (Issue November). The Queen’s University of Belfast. [van ’t Hoff - 2009.pdf](https://github.com/WEC-Sim/WEC-Sim/files/5119259/van.t.Hoff.-.2009.pdf)
[2] Yu, Y., & Jenne, D. (2018). Numerical Modeling and Dynamic Analysis of a Wave-Powered Reverse-Osmosis System. Journal of Marine Science and Engineering. https://doi.org/10.3390/jmse6040132 [Yu, Jenne - 2018.pdf](https://github.com/WEC-Sim/WEC-Sim/files/5119261/Yu.Jenne.-.2018.pdf)
[3] Yu, Y.-H., Li, Y., Hallett, K., and Hotimsky, C., 2014. “Design and Analysis for a Floating Oscillating Surge Wave Energy Converter,” In 33rd International Conference on Ocean, Offshore and Arctic Engineering, San Francisco, CA, United States. [Yu et al. - 2014.pdf](https://github.com/WEC-Sim/WEC-Sim/files/5119264/Yu.et.al.-.2014.pdf)

hossein-mehdipour, 2020-08-28:
Really appreciate your helps,
but for example, in afew OSWEC cases that I have been able to see these coefficients, the K value was 0, (as the example in the comment above) and I don't really understand if there is like a specific range for these values? Or I don't even know what is the alternative to this to be honest. Either there is a specific range, or it could be any numbers, or there is range but it is not known. I think the 3rd scenario is the case here.

dforbush2, 2020-08-28:
The K and C value referred to in the comment above (which I think you were referencing) are power take-off components. A PTO will passively add some stiffness and damping to the system, but these can additionally be thought of as active controller gains. For instance, a PI controller on flap angular velocity in the OSWEC device will end up acting like an additional damping (proportional term) and stiffness (integral term).

So with respect to the range of these parameters, there will be an optimal selection of gains based upon the frequency content of the exciting wave field, but not a practical analog bounding their range aside from perhaps the controller stability limits. I hope this is helpful.

dforbush2, 2020-09-02:
Within the WEC-Sim code, there is a lower limit of zero for both K and C.

closed 2020-09-16 by dforbush2