WEC-Sim/WEC-Sim #397: WEC Sim PDF??

sugarl14, 2020-08-16:
Hi,

I was wondering if there exists a PDF of the WEC-Sim documentation that is found on wec-sim's website? I found one online but its for wec-sim version 1.0.  I like to have a PDF available on my computer so I can make annotations and view it offline.

Thanks!

dforbush2, 2020-08-17:
At present, there is a not a PDF-compiled version of the documentation. However, I suspect you are not alone in this desire, and this should be doable in our current documentation approach, so I will see about adding one to the repository.

I will leave this issue open.

kmruehl, 2020-09-18:
@sugarl14 The WEC-Sim documentation is publicly accessible on the [WEC-Sim repository on the gh-pages branch](https://github.com/WEC-Sim/WEC-Sim/tree/gh-pages). There are a couple of ways you can take notes about the documentation and view it offline:

- You can look at the `*.rst` [source files](https://github.com/WEC-Sim/WEC-Sim/tree/gh-pages/source), which are used to compile the documentation and take notes there. 

- You can [compile the documentation as PDF](https://stackoverflow.com/questions/39534718/how-to-create-a-pdf-out-of-sphinx-documentation-tool) using the `make latexpdf` command. Here's documentation compiled as a pdf: 
[WEC-Sim.pdf](https://github.com/WEC-Sim/WEC-Sim/files/5242693/WEC-Sim.pdf)

closed 2020-09-18 by kmruehl