WEC-Sim/WEC-Sim #398: Error using wecSim (line 225)

Rodimus88, 2020-08-18:
i am new to use of wec-sim, i just try to execute examples given and got this issue. i have no idea of whats it means. it would be grateful to have a solution for this.

[image]

dforbush2, 2020-08-18:
Assuming you have an up-to-date WEC-Sim clone, this is a MATLAB-version compatibility issue related to the handling of variant sub-systems. More recent editions of matlab can handle structures in variant expressions, apparently 2016a cannot. I'll make this change on the source code to address this compatibility issue. 

As a work-around in the mean time, try replacing lines 225 through 228 of wecSim.m with:

```
sv_regularWaves=Simulink.Variant('typeNum>=10 && typeNum<20 && yawNonLin~=1');
sv_regularWavesNonLinYaw=Simulink.Variant('typeNum>=10 && typeNum<20 && yawNonLin==1');
sv_irregularWaves=Simulink.Variant('typeNum>=20 && typeNum<30 && yawNonLin~=1');
sv_irregularWavesNonLinYaw=Simulink.Variant('typeNum>=20 && typeNum<30 && yawNonLin==1');
```

Rodimus88, 2020-08-18:
i shift from WEC-sim v.4.1 to WEC -sim v2.1, and it works fine. but i had to run the bemio.m first before run >>wecSim. but i am satisfied with it.

Rodimus88, 2020-08-18:
your solution has worked perfectly. thank you again for the help. when running 
[image]
the examples in the WEC-sim up-to date clone, lot of warnings did pop up. but it does not occur in the WEC-sim v2.2. still results are generated. can you tell me how to avoid those warnings.

dforbush2, 2020-08-18:
Those xml files are generated by simulink at run time, the exact format of which is particular to a matlab/simulink version. I do not see these warnings in WEC-Sim 4.1 running on 2020a or 2019a, so I suspect this has something to do with 2016a. 

As long as the results look reasonable they shouldn't present a problem, but I am not sure how to suppress them.

Rodimus88, 2020-08-19:
Thank you for your insight and help.

closed 2020-08-19 by Rodimus88