WEC-Sim/WEC-Sim #399: Mooring line slacks

eirinikatsido, 2020-08-19:
Hello!

I am trying to simulate an point absorber wec, please check the link for the description of the operating principle (https://user-images.githubusercontent.com/44980936/90624078-afd02380-e217-11ea-84c5-8c4b070fa410.png). 

When the lower endstop spring inside the PTO is fully compressed,  the translator sits on the lower position (during a wave trough) and the chain, connecting the PTO with the buoy, slacks. It  means that the force from the PTO system transferred to the buoy through the connection chain is equal to zero. In other words, the motion of the buoy and the translator becomes uncoupled! My question is how to define the slack of the mooring line in wec-Sim?

I have created a Simulink model considering only the heave motion (even though the system has 6DoF). In that model the buoy is connected with the Translational PTO with actuation force. The latter is coupled with a Subsystem defining the weight of the translator as the actuation force. Moreover, in the settings of the Translational PTO the Upper and Lower limits have been activated in order to define the stroke length of the PTO. 

So the force transferred to the buoy through the chain is: 
FconnectionChain = -massTranslator*g + Fpto + Fspring, but when the lower spring is fully compressed FconnectionChain = 0. 

Best Regards, 
Eirini

yuyihsiang, 2020-08-19:
HI Eirini,

The short answer is that WEC-Sim can not handle a slack mooring directly. 

Having said that you should be able to modify the way you calculated the actuation force so that the chain force becomes zero when the chain slacks. If you want, you can upload your files, including the hydrodata and the models, and I can take a look at it.

--YI-Hsiang

eirinikatsido, 2020-08-20:
Hello YI Hsiang, 

Thank you for your reply! Here kindly attached my case.

In the subsystem there are the forces acting on the PTO for the several stages, depending on the translator position: i) when the upper end-stop spring is compressed, iii) when the upper end-stop spring is fully compressed and iii) when the connection chain slacks. It the latter case I applied what you described in your previous post, the forces acting on buoy due to the PTO are canceled by adding the same but opposite sign forces, which are the translator weight and the Fpto.

Best Regards, 
Eirini
[PA_Heave.zip](https://github.com/WEC-Sim/WEC-Sim/files/5103969/PA_Heave.zip)

yuyihsiang, 2020-08-24:
Hi Eirini,

Thank you for uploading the files. 

Just to check, you mentioned that "It the latter case I applied what you described in your previous post, the forces acting on buoy due to the PTO are canceled by adding the same but opposite sign forces, which are the translator weight and the Fpto." Did that solve your issue? 

--Yi-Hsiang

eirinikatsido, 2020-08-26:
Hello Yi, 

By applying the forces cancellation by adding the same and opposite force, it seems that the PTO does not interact with the buoy. A brief comparison of the wecSim solution with the results from OpenFOAM simulations show relatively good agreement in terms of connection chain force. 

Best Regards, 
Eirini

yuyihsiang, 2020-09-16:
I am going to close the issue. Let me know if you have any questions and I can re-open it if needed.

--Yi-Hsiang

closed 2020-09-16 by yuyihsiang