WEC-Sim/WEC-Sim #400: How to create hydrodata .h5 file

Rodimus88, 2020-08-19:
can any one give guidance in creating a custom hydrodata file.

dforbush2, 2020-08-19:
Try this to start.
https://wec-sim.github.io/WEC-Sim/advanced_features.html#writing-your-own-h5-file

Rodimus88, 2020-08-20:
thank you for the help..!!

Rodimus88, 2020-08-20:
Does any one know how to create hydrodynamic data from the Ansys AQWA solver which can be use to run in BEMIO to genertae .h5 files.?

dforbush2, 2020-08-20:
With regard to reading AQWA output files, see *./examples/BEMIO/AQWA/Example.

If you are asking how to set up an AQWA run, that is not something I can help you with much. The structure of AQWA outputs seem to be a strong function of user selected solver configurations, which makes writing a reliable parser within BEMIO difficult. I would recommend that you reference the AQWA output files included in the example mentioned and attempt to match their formatting.

Rodimus88, 2020-08-20:
thanks you dforbush2 for the help.!!

Rodimus88, 2020-08-20:
I encounter with two errors when running the bemio.m file in the *./examples/BEMIO/AQWA/Example.
[image]

Rodimus88, 2020-08-20:
I come up with two errors in  running bemio.m file in /examples/BEMIO/AQWA/Example.
[image]

dforbush2, 2020-08-20:
Ah its an easy one! 
Try replacing 

`TF  = startsWith(raw1(n),'*');`

with 

`TF  = startsWith(raw1{n},'*');`

(note the change to curly braces { } around n)

Rodimus88, 2020-08-20:
I have done the change, but the error still pops up.
[image]

dforbush2, 2020-08-20:
I'm sorry that didn't work. It's hard for me to debug this exactly without a 2016a version. Assuming this fails on the first iteration, and not some later line, the problem is startsWith wants a string and in 2016a (n) gives you a cell and {n} gives you a character array. 

There are a number of data-type conversion functions like 'string', 'convertCharsToStrings' that I would investigate. I would throw a debug point on line 49, and then experiment a little with things like

`TF = startsWith(string(raw1{n}),'*');`

but again I can't fully debug this since I'm running different matlab versions.

Good luck!

dforbush2, 2020-08-20:
In lieu of startsWith, I would check out using regexp with a logical expression. Something like

```
startIndex = regexp(raw1{n},'*')
if startIndex == 1 
TF = 1 ;
else TF = 0;
end

```