WEC-Sim/WEC-Sim #401: varying parameters, power plots

dforbush2, 2020-08-19:
Thanks you helped me resolve all problems. Its fine now.
Now i have 2 questions
1. I need to know another thing which is can i vary the wave height , time period and damping coefficient in the code given.
2. how to get or plot power generated graph in the example model which i simulated.

Thanks.
[image]

_Originally posted by @Carbon360 in https://github.com/WEC-Sim/WEC-Sim/issues/395#issuecomment-676396563_

dforbush2, 2020-08-19:
To run parameter sweeps of wave height, period, or damping coefficients, check out Multiple Condition Runs https://wec-sim.github.io/WEC-Sim/advanced_features.html#multiple-condition-runs-mcr. There is also a webinar for your reference.

To plot PTO power, you can specify this as a desired output in your userDefinedFunctions.m file that is located in the same directory as the simulink *.slx file you are running. The output variables in the response class will be useful here. https://wec-sim.github.io/WEC-Sim/api.html#response

Carbon360, 2020-08-20:
Thank you so much...
i saw some codes like this below...

%Plot heave forces for body 2
output.plotForces(2,3);

But how do i plot for power 
can you please give example of code that must be used. PLease

dforbush2, 2020-08-20:
Because there are a number of ways to define a power for a WEC this is difficult to answer in general. For the WEC-Sim/examples/OSWEC, the mechanical power is calculated from the damping and stiffness values specified in the PTO. These end up in the output.ptos.powerInternalMechanics field. The device is free in pitch (5), so power can be plotted:

figure; plot(output.time, output.ptos.powerInternalMechanics(:,5));

But again I emphasize the appropriate power to consider will depend on your model structure and your applications.

dforbush2, 2020-08-24:
My mistake. There is no output.time field. But there are output.ptos.time, output.constraint.time, among others. 

Try exploring the output structure for field names in matlab and in the documentation. http://wec-sim.github.io/WEC-Sim/api.html#response

closed 2020-09-02 by dforbush2

abhisheks865, 2020-11-19:
I am sorry but just to be clear. Can you please let me know the code should be used to get same available vs electrical power output chart as above for RM3 model where heave is the main force to generate the power?