WEC-Sim/WEC-Sim #402: Restoring forces and Added Mass forces comparison

Liam-Guerrero, 2020-08-19:
Hi Guys,

I wonder if you could give me a hand to understand how wecSim computes some of the forces/moments.

I am working with the scaled-wavestar-like model developed for the WECCCOMP and ran the model for regular wave, height 0.06 m and period 1.2 s. 

Now, I want to compute the restoring force and the added Mass force, for which I believe I have to use the pitch values so:

1. For the restoring force, I take the position ( **output.bodies(1,1).position(:,5)** ) and multiply this vector by the hydrostatic coefficient ( **body(1,1).hydroForce.linearHydroRestCoef(5,5)** )

However when I compare this vector with the restoring force from **output.bodies(1,1).forceRestoring(:,5)** the values are way different from each other. Not just in magnitude but also there is an offset between the signals.
[image]

The red figure is the one I compute by "hand" and the black one is the restoring force output from wecSim.

2. Similar to the restoring force, for the added mass force, I take the acceleracion ( **output.bodies(1,1).acceleration(:,5)** ) and multiply this vector by the infinitive added mass (which I believe is  **body(1, 1).hydroForce.fAddedMass(5,5)** )

However when I compare this vector with the restoring force from **output.bodies(1,1).forceAddedMass(:,5)** the values are way different from each other. In this case, the difference is in magnitude but also there is a phase difference in 180 degrees.
[image]
The red figure is the one I compute by "hand" and the black one is the added mass force output from wecSim.

I really appreciate any help to understand why my calculations are not similar to the ones from wecSim.

Attached are the wecSimInputFile used for the simulation, the WaveStar_matlabWorkspace.mat and the script I created for the calculations of the forces.

Kind Regards, 

Juan Guerrero
[forcesCalculations.zip](https://github.com/WEC-Sim/WEC-Sim/files/5099177/forcesCalculations.zip)

nathanmtom, 2020-08-21:
@Liam-Guerrero Thank you for your question and happy to try and answer your question. 

To answer your first question, the hydrostatic restoring force time history for the WECCCOMP is not simply the **body(1,1).hydroForce.linearHydroRestCoef(5,5)** * **output.bodies(1,1).position(:,5)**. Rather the hydrostatic coefficient matrix has coupling terms with surge and heave since in this example the center of buoyancy and the center of gravity are not aligned at the same X and Y coordinates. In addition because the center of buoyancy and the center of gravity are not aligned in the X-Y plane there is also a constant overturning moment that also needs to be accounted for in the calculation. I've updated the forceCalcuations.m file you provided in the attached .zip file to account for all moments generated in the hydrostraic force and moment calculation. Please review and let me know if you still have additional questions. 

To answer your second question, this is similar to the first question where the equation for the added mass force is not simply the pitch added mass in the (5,5) index multiplied by the pitch acceleration. Here again, there are coupling coefficients with the surge and heave motion and you will need to account for in the moment calculation. Please see the updated forceCalcuations.m and let me know if you have further questions.

I've also attached a plot below now showing agreement between the WEC-Sim output and the theoretical calculation. 

[image]

[forcesCalculations_R1.zip](https://github.com/WEC-Sim/WEC-Sim/files/5106088/forcesCalculations_R1.zip)

Liam-Guerrero, 2020-08-26:
Hi Nathan,
Thanks for your help, it definitely helps me.

I still have some open questions, if you do not mind.

I understood about the constant overturning moment since the centre of buoyancy and the centre of gravity is not aligned at the same X and Y coordinates. But still, I have an open question, the fact that the hydrostatic coefficient matrix has coupling terms with surge and heave is due to the fact that the float oscillates in translation in x, z-axis and rotates about y-axis. I mean, even if the cob and the cog were aligned, the hydrostatic coefficient matrix would have coupling terms. right? 

Also, there is this structure body(1).hydroForce.store, which contains the fields of mass, momOfInertia and hydroForce_fAddedMass, and checking the Linear Hydrostatic Restoring Force subsystem in Simulink this body.hydroForce.storage.mass is used instead of body.mass. 
[image]

Later, checking the bodyClass file, I see how it is computed, within a function called adjustMassMatrix:
                tmp.fadm=diag(obj.hydroForce.fAddedMass);
                tmp.adjmass = sum(tmp.fadm(1:3))*adjMassWeightFun;
                obj.mass = obj.mass + tmp.adjmass;
But still do not the meaning of this mass. Why is this mass used?

nathanmtom, 2020-08-27:
@Liam-Guerrero Appreciate the follow-up question and will attempt to answer your questions. 

First, I must apologize as my first response was not completely correct. For the hydrostatic restoring coefficient matrix, there is no surge-pitch coupling term, but there is a heave-pitch coupling term. I misspoke and also need to clarify that you are correct that the coupling terms are not associated with the misalignment of the center of buoyancy and the center of gravity. Rather the heave-pitch coupling term will be present depending on the shape of your hull and where the center of gravity is defined. As per the WAMIT calculation for the heave-pitch coupling term, shown here: 

[image]

This equation will boil down to an integration of the waterplane area of the hull that has been multiplied by the level arm '_x_'. If the body is symmetric and is centered in the x-y plane at (0,0), then the '_x_' term will be asymmetric and when performing the integration will result in a zero value. If the body remains symmetric and centered at (0,0), but we shift the center of gravity forward (as in the WECCCOMP) then '_x_' is no longer symmetric and the integration should result in a nonzero value. 

To your second question, the numerical solver in MATLAB/Simulink does not allow the calculation of the added mass force to be on the left hand side of the equation of motion (F=m*a). Therefore, WEC-Sim shifts the largest added mass along the diagonal to the right hand side of the equation of motion. But, certain terms such as the buoyancy, that  you have shown in your figure, relies on the material mass of the body and during the simulation we store the material mass of the body in the body.hydroForce.storage.mass variable and then correct after the simulation is done with the adjustMassMatrix function. 

Please let me know if this has answered your questions. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

nathanmtom, 2020-09-02:
@Liam-Guerrero I wanted to follow-up to see if my latest response has answered you questions? 

Cheers, 
Nathan
National Renewable Energy Laboratory

Liam-Guerrero, 2020-09-02:
Dear Nathan,

Yes, sure it really helped me to clarify my doubts regarding the forces. I still have some issues to understand the storage mass, but I think this is more about Matlab/Simulink solver, so I'll leave it aside for the time being.

I also have some open questions about how to extract the matrices for the state-space model for the radiation force/moment and one regarding the excitation force, but I think I'll post a new issue.

Once again, thanks for your help.
Kind Regards
Juan Guerrero