WEC-Sim/WEC-Sim #403: Hydrostatic stiffness for pitch mode

Liam-Guerrero, 2020-08-19:
Hi Guys, 

I wonder if you could give me a hand to understand how wecSim computes Hydrostatic stiffness for pitch mode.

I am working with the scaled-wavestar-like model developed for the WECCCOMP and ran the model for regular wave, height 0.06 m and period 1.2 s.

Now, I want to compute the restoring moment using the linear equation Mhd(t) = −K_5*θ(t).

First, I had understood that the hydrostatic stiffness for pitch mode is computed as: 
K_5 = rho * g * Iyy, and for the wavestar-like model would be: **K_5=rho * g * body(1,1).momOfInertia(2) = 14.58 Nm/rad**;

However, if I extract the value from the hydroForce structure (which bodyClass.m file indicate  % Hydrodynamic forces and coefficients used during simulation.) **body(1,1).hydroForce.linearHydroRestCoef(5,5)** is equal to 0.2175, which I think has the units of Nm/rad.

I searched a bit more about this topic and found a Paper [Mooring and Hydrostatic Restoring of Offshore Floating Wind Turbine Platforms](https://ieeexplore.ieee.org/stamp/stamp.jsp?arnumber=7003269) in which it is explained how to compute the Hydrostatic Restoring coefficients. From which I extracted the following figures:
[image]

In the same paper, they present the hydrostatic stiffness matrix, which confused me even more because they compute a K_pitch and additionally a K_55, which I first thought are the same, but they are different. I even computed the K_55 according to the paper but the value is far from the value used by wecSim: 

`rho=1025;
    g=9.81;
    A=0.25*pi*0.254^2;
    W=body(1,1).mass;
    zg=body(1,1).cg(3);
    zb=body(1,1).cb(3);
    Fb=body(1,1).dispVol * rho * g;
    Kp=rho * g * body(1,1).momOfInertia(2)
    Kh=rho * g * A
    xf=0.5487; 
    
k55=-W*zg+Fb*zb+Kp+Kh*xf^2`
 k55=1.664560999380456e+02

So in summary, I would like to know how the hydrostatic stiffness for the pitch mode (which I think is the one we should use for this model) and how wecSim computes it.

All these topic of hydrodynamics forces/coefficients is new for me, and I have not had any course on my university curriculum before. So I really appreciate any guidance to understand how to compute these coefficients.

Kind Regards
Juan Guerrero

nathanmtom, 2020-08-20:
@Liam-Guerrero Thank you for your question and we are happy to try our best to answer your question. 

To your first question, it appears that you have misinterpreted I_{yy} as the body pitch moment of inertia when instead it should be the [second moment of area ](https://en.wikipedia.org/wiki/List_of_second_moments_of_area#:~:text=The%20second%20moment%20of%20area,regard%20to%20an%20arbitrary%20axis.&text=If%20the%20piece%20is%20thin,the%20area%20moment%20of%20inertia.) of the water plane area. If you checked the dimensions of the equation for K_{5} you should see that it does not end up with Nm/rad. 

For the WECCCOMP model, the hydrodynamic restoring coefficients were calculated using WAMIT and I will refer you to their user [manual](http://www.wamit.com/manualv7.3/wamit_v73manual.html), see Section 3.1, for the derivation of the K_{55} hydrostatic restoring coefficient. 

To your question about the derivation from your linked paper, you should first replace I_{yy} with the second area moment of area and check your answer. After reviewing the equation you have listed with the WAMIT definition, I believe the two equations are the same. The expression in WAMIT is more general and if you were to complete the integral calculation, written for C(5,5), you would get the expression from your paper when using the parallel axis theorem to calculate the second area moment of area.

If you would like to see more information on how the hydrostatic and gravitational forces are calculated in WEC-Sim, you can look in the Hydrodynamic Body Block -> Hydrostatic Restoring Force Calculation -> Linear and Nonlinear Restoring Force Variant Subsystem -> Linear Hydrostatic Restoring Force.

I hope these responses have helped answer your question, but please let us know if you are still having issues on this topic. 

Cheers, 
Nathan

Liam-Guerrero, 2020-08-26:
Hi Nathan, 

Thanks for your support, it really helped me a lot.

I still have some questions regarding the linearHydroRestCoef matrix.

body(1).hydroForce.linearHydroRestCoef =
0	0	0	0	0	0
0	0	0	0	0	0
0	0	519.33	-0.6095	25.8771	0
0	0	-0.6095	-1.0710	-0.0303	0
0	0	25.8771	-0.03036	0.21749	0
0	0	0	0	0	0

If I compared some of those values with the WAMIT definition, they do not match:
[image]

See for example:

C(3,3) = rho* g* Ac = 1025* 9.81* 0.25* pi* (0.256)^2 = 517.56, which is different from the body(1).hydroForce.linearHydroRestCoef(3,3), let's say that is maybe due to round errors.

But 
C(4,6) 
[image]

which from the body(1).hydroForce.linearHydroRestCoef(4,6) it could be either that xf=0 or that rho* Vo = m, however xf is not zero, neither the second option:

0.00344635* 1025 = 3.5325, which is different from the mass of the float = 3.075

same for C(5,6).

Also for: 
[image]

from the paper K(3,4) :
[image]

If this equation is compared with body(1).hydroForce.linearHydroRestCoef(3,4) I could be interpreted as if there is a lever arm in the y-direction. However, I could not understand the meaning of this yf 

yf = body(1).hydroForce.linearHydroRestCoef(3,4) / body(1).hydroForce.linearHydroRestCoef(3,3) = -0.0012
 
Finally, I could not compute the K55 as the discretisation presented on the paper, Is there any chance to some help with this?

Kind Regards, 
Juan Guerrero

nathanmtom, 2020-08-27:
@Liam-Guerrero Thank you for the following on question and I will do my best to respond to your questions. 

First, please note that in this model the water density was originally set at 1,000 kg/m^{3}. The model was built to represent an experimental apparatus that was tested in a fresh water tank and the initial mass of the float was determined assuming this.  

Second, the C(3,3) term is slightly different from your theoretical derivation. In WAMIT, the various integrations are completed over a mesh of the floating body and at times the mesh resolution may result in a slightly different value.

For the C(4,6) term, in our calculation we have used the Alternative Form 1 of the FRC file, so WAMIT will assume that the mass and displaced mass are equal. You are correct in pointing out that there is a discrepancy between the material mass in the wecSimInputFile.m and the displaced volume from the .h5 file, but the WAMIT calculation assumes they are equal.

[image]

For C(3,4), based on the paper you have listed y_{f} is the y-coordinate of the platform weight. I've pulled out the excerpt from the paper below:

[image]

For C(5,5), I have followed the equation giving in the table listed in your response and have obtained a value close to that reported from the WAMIT output; however, I did have to adjust the diameter at the water plane area slightly from 0.256 m to 0.2549 m). This feeds off my point about C(3,3) where the numerical integration of the provided mess may differ from the input values. 

I hope these responses have answered your questions, but please let me know if you still have concerns or additional questions. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

Liam-Guerrero, 2020-09-02:
Dear Nathan, 

Thanks for your help, It really helped me to clarify my doubts.

Kind Regards
Juan Guerrero

closed 2020-09-02 by Liam-Guerrero