WEC-Sim/WEC-Sim #405: Automatic Object Creation

H0R5E, 2020-08-25:
Hi,

I have found a method for automatic creation of objects (such as a bodyClass) when setting the mask parameters on the WEC-Sim simulink blocks. Basically, you can create a callback to read the parameter name and then add the object to the base workspace. For the bodyClass, the code looks something like this (note, I renamed the default value to `new_body`):

```
value = get_param(gcb,'body');

if ~strcmp(value, "new_body")
    setvar = sprintf('%s = bodyClass("");', value);
    evalin('base', setvar);
end
```

My motivation for this is to reduce the need for users to understand initialization of classes, as the correct class will now be added to the workspace automagically. If this is something that you think would be useful I can put in a pull request using the bodyClass as an example. If I did that, would you need me to revert to an older version of MATLAB?

Cheers,

Mat

kmruehl, 2020-09-18:
@H0R5E This is great! Thank you. I added it to the project board as an update. It may be a bit before we merge the PR, but if you would like to draft one, that would be great.

H0R5E, 2020-10-01:
Hey @kmruehl,

So, after doing a bit more work on this I realise there is currently no point to it, given that any objects created in the base workspace are overwritten when wecSim in called. Given this is the case, the next best thing would be to stop the blocks looking for objects in the first place, which is possible, but the change to make this work would break existing models.

I think if we were going to break existing models with a change, then it should be for a more ambitious one. I think the UI of wecSim could be improved significantly by simply making it possible to use the run button in the simulink interface. This should go alongside a change to the way wecSim is configured to either:
    1. Use objects defined in the workspace as inputs
    2. Use a configuration file (such as an ini) file to configure blocks

The first option would allow more dynamic operation, in particular, the strict naming of objects that is required now would be relaxed. The second option would reduce the need for users to understand classes and is most similar to the way wecSim is currently used.

So, yeah, let me know if any of this is of interest to the WEC-Sim team. I have a few other thoughts about improving the code / docs organisation, but perhaps that is best discussed in another thread. 

Cheers,

Mat

kmruehl, 2020-10-26:
@H0R5E after some discussion we would to keep the wecSim execution the same, but will consider adding an optional parameter that allows users to use the SIMULINK interface.

closed 2020-10-26 by kmruehl