WEC-Sim/WEC-Sim #406: Defining userDefinedFunctions.m

Rodimus88, 2020-08-26:
how to set up the userDefinedFunctions.m file for a custom problem. i just copy the userDefinedFunctions,m file from OSWEC tutorial and place in my project folder. then i encounter with following error,
[image]

nathanmtom, 2020-08-27:
@Rodimus88 You are correct that if a userDefinedFunction is in the same working directory as your WEC-Sim files at the end of the simulation WEC-Sim will call the userDefinedFunction. In your screenshot above it appears that the userDefinedFunction has been called successfully, but it appears that there is no 'output' object in your 'Workspace'. Therefore, the userDefinedFunction stops and does not run to completion. You will need to complete a WEC-Sim run to generate the output object. 

Please let us know if this solves your issue. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

Rodimus88, 2020-08-28:
your solution has worked Mr @nathanmtom . thank you for your help.!!

closed 2020-08-28 by Rodimus88