WEC-Sim/WEC-Sim #407: rigit,bottom-mounted, fixed vertical wall in the simulation

eleftheria-pap, 2020-08-28:
Hello,

I would like to implement a time domain analysis of point absorbers in front of rigid, bottom-mounted, fixed vertical wall using WEC-sim. I haven't used WEC-sim before and I would like to ask you if this problem could be simulated in WEC-sim. Especially, I would like to know if the wall, representing a fixed boundary interacting with the point absorbers, could be considered in the simulations. The interaction with the wall would be considered in the BEM tool to calculate all required inputs for WEC-sim.

I am looking forward to your response.

Thank you very much in advance.

Best Regards,
Eleftheria

dforbush2, 2020-08-31:
Hi Eleftheria,

Assuming you run the BEM simulation with the wall geometry as well, then you can run WEC-Sim with body-to-body interactions on. This will include the BEM data describing forces due to the interactions between the point absorber and the wall in a linear fashion.

https://wec-sim.github.io/WEC-Sim/advanced_features.html#body-to-body-interactions.

eleftheria-pap, 2020-09-01:
Thank you for your response. 

The BEM model will be indeed run taking into account the presence of the wall. In the BEM simulations the wall is considered to be FIXED and bottom mounted. So, for the wall there is no radiation damping and/or added mass coefficients. The radiation damping and added mass of the WEC are caclulated in the BEM simulations taking into account the presence of the wall boundary. Accordingly, the augmented matrix (6x2, 6) will have NON-ZERO elements ONLY for i,j=1,...,6. Exciting forces do exist for both the WEC and the wall. 

I guess WEC-sim could consider the above case using the bob-to-body interaction module.

I will start working on the modelling of the problem in WEC-SIm, but I would like to confirm with you the above.

Best Regards,
Eleftheria

nathanmtom, 2020-09-02:
@eleftheria-pap Thanks for the follow up comments. Our team agrees with you that when running your BEM solver, if the wall is held fixed then in simulations there will be no radiation forces or body-to-body interactions with the oscillating WEC. Depending on the BEM solver used and settings used, there still may be non-zero radiation and added mass coefficients for the wall, but since it will be fixed in the WEC-Sim run it would be the same as oscillating with zero radiation and added mass coefficients. 

Our team is confused by the augmented (6x2,6) matrix you have listed, as any two body system would have an added mass and radiation coefficient matrices that would be (6x #of bodies, 6 x #of bodies); however, perhaps you are referring only to the coefficients associated with the oscillating body, assuming the WEC is listed as body 1 in the BEM solver? 

In WEC-Sim, if body-to-body interactions are turned off, then the only hydrodynamic radiation forces would be associated with the oscillation of each body on itself and there would be no coupling terms. In your case, with a two body system where one body is fixed should provide the same results as not activating body-to-body interactions since the second body won't move. However, we have assumed that the main results of interest would be the response of the WEC but if you wanted to estimate the forces on the wall given the WEC motion then body-to-body interactions would need to be activated. 

Please let us know if the response above confirms your understanding of WEC-Sim's operations. 

Cheers, 
Nathan 
National Renewable Energy Laboratory

eleftheria-pap, 2020-09-07:
Thank you Nathan for your detailed response. At this stage I think everything is clear to me. I will start modelling the problem and I will come back to you in case I have any further questions.

Thank you once more.

Best,
Eleftheria

closed 2020-09-08 by nathanmtom