WEC-Sim/WEC-Sim #409: Pitch Response Graph in userDefinedFunctions.m

Rodimus88, 2020-08-30:
I ran a custom made floating sphere in wec-sim. in the grpahs output by userdefinedfunction.m I recieved a blank graph for the float pitch response. what is a reason for this. what should i do to get responses like velocity and position mentioned in that graph. 
[image]

dforbush2, 2020-08-31:
Hi @Rodimus88 

There is an *.h5 file for a sphere in the WEC-Sim applications repository (*./Free_Decay) that may be useful as a check. 
Is the sphere constrained via a 3 or 6 DOF constraint, which allows pitch motion? Depending on your excitation frequency, there may not be reason to expect pitch excitation in the sphere. 

WEC-Sim applications : https://github.com/WEC-Sim/WEC-Sim_Applications

Rodimus88, 2020-09-01:
I had used translation constrain in this situation. so its my fault. thanks @dforbush2 for pointing it out. anyway is there a way to collect responses as above of a translating body using userdefinedfunction.m ?

dforbush2, 2020-09-01:
I would recommend you check out the response class, which holds the simulation output data. Most of the example userDefinedFunctions.m scripts are merely plotting the variables contained in the `output` structure.

http://wec-sim.github.io/WEC-Sim/code_structure.html#output-structure
http://wec-sim.github.io/WEC-Sim/api.html#response

These response/force data are usually presented in 6 DOF [surge, sway, heave, roll, pitch, yaw]. So in your example, if we assume thee degrees of freedom something like:
```

figure; clf;
plot(output.bodies(1).time, output.bodies(1).position(:,[1,3,5]))
```

should work for you.

Rodimus88, 2020-09-02:
It works, thank you @dforbush2 for your insight...!! i tried sphere from BEM data in the WEC-Sim applications repository (*./Free_Decay/hydroData) you mentioned in a previous comment. there was an error in it. i think it may be error due to my version of matlab. can you give a solution for this. here is the error i receive.
[image]

dforbush2, 2020-09-03:
Hi @Rodimus88 

I agree this is a matlab version issue. I do not have a 2016a version to test this on, but in my experience this issue is likely from the change over the last few Matlab versions in how Simulink handles one-dimensional arrays vs. row and/or column vectors (i.e., a signal dimension of (6) vs. a signal dimension of (6 x 1) or (1 x 6). While the behavior of `circshift` did change in 2016b, it seems like the syntax in the function should still work in 2016a...but from the error you posted I'm not sure if the simulink error (related to port dimensions) caused the function call to fail, or vice-versa. 

I would try to add a reshape block the output of the selector to make the input to v explicitly (N x 1) column vector, or (1 x N) row vector, or a 1-D array of length N, (try all three, these are dialog box options within the reshape block). You'll need to unlock the library-linked body block to make this modification, but my idea is to make this variant subsystem of the body block look like this, where I've highlighted the proposed added block. This screenshot is of the sphere/Rigid Body/Hydrodynamic Body/Wave Radiation Forces Calculation/SS CI and Constant-Damping-CoeVariant Subsystem/Convolution Integral Calculation subsystem.

[image]

Rodimus88, 2020-09-03:
Thank you @dforbush2 for your consideration !. I hope to switch from matlab 2016 to 2019 for ease in any trouble in future problems. if i was unable to upgrade to 2019 version for any reason. i will try this solution and will give you a feedback. thank you again for the help given.!

closed 2020-09-03 by Rodimus88