WEC-Sim/WEC-Sim #411: State-Space model to reconstruct the wave excitation/radiation moment for WECCCOMP apparatus

Liam-Guerrero, 2020-09-03:
Hi Guys, 

I need to build an equivalent equation-of-motion model for the wavestar-like device used in the WECCCOMP. 

1. State-Space model to reconstruct the wave excitation moment w.r.t the hinge A, to use it as input to the device model: I am planning to incorporate non-linear hydrostatic force considering the free surface elevation eta and the device displacement zd(t), so I would like to estimate/predict the free surface elevation instead of the wave excitation force/moment. 
If I am not wrong, using the output from wecSim, the wave excitation moment w.r.t point A can be computed as: 
M_ex = F_ex,x * sin(theta) * l_arm - F_ex,z * cos(theta) * l_arm + M_ex,pitch.

However, as mentioned above I would like to have something like:
\dot(p) = A_ex * p(t) + B_ex * \eta
M_ex = C_ex * p(t) + D_ex * \eta

Any clue how to proceed with this "conversion"?

2. State-Space model to compute the wave radiation moment: Similar as the above I want to be able to compute as:
\dot(r) = A_r * r(t) + B_r * \dot(\eta)
M_r = C_r * r(t) + D_r * \dot(\eta)

I have seen from the model built in wecSim the matrices for the SS, body(1).hydroForce.ssRadf. However I am not sure how to extract the right values, and combine them to model the wave radiation moment w.r.t the hinge A.

I hope this is not a long issue, and some can help me.

Thanks in advance.

Cheers, 

Juan Guerrero

nathanmtom, 2020-09-08:
@Liam-Guerrero Thank you for your question and I'll do my best to try and answer. 

To your first point, one difficulty is that the wave-excitation force values are noncausal which means that you theoretically need the wave elevation profile for a certain time in the past-to-future to calculate the wave excitation force at time _t_. A very good reference can be found [here](https://www.sciencedirect.com/science/article/pii/S0141118796000077). Therefore, to obtain a state-space representation you would need to implement a causal time delay, refer to [this reference](https://www.sciencedirect.com/science/article/pii/0141118796000028), or make a causal approximation using a system identification approach. Here you could simulate WEC-Sim under a series of random waves and generate a transfer function between M_{ex}/\eta, about the hinge A, and fit a linear state space model to the frequency response and suggest reviewing [this reference](https://www.sciencedirect.com/science/article/pii/S0029801807002363). In fact, one of the competitors in WECCCOMP used a similar approach to build a state space model to develop a model predictive control strategy and described their approach in a [conference paper here](https://asmedigitalcollection.asme.org/OMAE/proceedings/OMAE2019/58899/V010T09A036/1068231). 

To your second point, it would be similar to my answer above where you

Liam-Guerrero, 2020-09-09:
Dear Nathan, 

Thanks for your help. Actually I just read the paper from Bradley A. Ling (the conference paper you mentioned) and was planning to do something similar. 

I still have a couple of open question, that if you do not mind, I would like to ask here:

1. Can you help me to confirm if, using the output from wecSim, the wave excitation moment w.r.t point A can be computed as:
M_ex = F_ex,x * sin(theta) * l_arm - F_ex,z * cos(theta) * l_arm + M_ex,pitch.

2. For the wave radiation moment, I could simply use the SS matrices given in the calling paper for WECCCOMP. However, I would like to understand the dimensions of the matrices given in the Numerical model for the WEC Control Competition (WECCCOMP) using WEC-Sim to model the WaveStar. Where the dimensions are:
size(body(1).hydroForce.ssRadf.A) = 96x96
...B = 96x6
...C=6x96
...D=6x6

Here I would guess the 6 is the DoF of the body, but no clue how to read the 96 and the most important, how to extract from that information the right values to built the ss-model for wave radiation force/moment for any particular dof, let's say heave for example.

3. And last but not least important, I ran the wecSim model for RW with waves.H=0.06 and waves.T=1.2, and I found counterintuitive results regarding the float vertical displacement compared to the wave amplitude, 

eta    = output.wave.elevation;
time = output.wave.time;
position=output.bodies(1).position.'-[body(1).cg;0;0;0];
verticaldisplacement = position(3,:);

plot(time(1:end/2,1),verticaldisplacement(1,1:end/2),'k',time(1:end/2,1),eta(1:end/2,1),'r--')
legend('Float Vertical Displacement','Wave Elevation')

[image]

How is possible that the float vertical displacement is bigger the wave elevation amplitude and the mean is slightly below the mean of the wave elevation?

Cheers, 
Juan Guerrero

nathanmtom, 2020-09-11:
@Liam-Guerrero Thank you for your follow-up comments on this issue. I'll do my best to answer your question. 

1. Without having completed a derivation, your equation looks correct; however, I would suggest that replacing the rotational constraint at Hinge A with a fixed constraint, then after running WEC-Sim you can plot the rotational constraint force against your analytical equation to see if they are equal. 

2. First, did you rerun BEMIO with different options for the state space implementation of the radiation coefficient? The .h5 included in the WECCCOMP has 324 states rather than 96. Regardless, each degree of freedom of motion may have a different number of state space states required to adequately represent the convolution integral calculation.

The WEC-Sim variable that includes the number of states for each degree of freedom can be found here: 
      - body(1).hydroData.hydro_coeffs.radiation_damping.state_space.it

The numbers each cell of the [6x6] matrix will correspond to the number of states for that degree of freedom. If you were to print body(1).hydroForce.ssRadf.A to the MATLAB Command Window you should see the matrices all lie along the diagonal. The way the matrices inputted into the large A matrix will start with the [1,1] index, move to the [1,2] , then [1,3] all the way till the [6x6] index. Knowing the degree of freedom of interest and the size of the state space matrices for each index should allow you to extract the appropriate state space matrices as needed. The same process can be used to extract the B and C matrices.

3. Yes, it is possible to have a float vertical displacement that is bigger than the wave amplitude. For example, point absorber WECs are often designed to resonant at a desired wave frequency which is associated with amplitudes of motion much larger than the driving input. There is likely a small mismatch between the defined mass of body(1) [the float] + body(2) [the arm] and the displaced volume of body(1) [the float]  which is causing the float to sink about ~1 cm. The WECCCOMP model parameters where chosen to fit a set of experimental data and therefore there is the possibility that there might be some slight offsets that one would not expect to see in a perfectly balanced system. 

I hope my responses have answered your questions, but if not, please feel free to ask further questions or clarifying comments.

Cheers, 
Nathan
National Renewable Energy Laboratory

Liam-Guerrero, 2020-09-15:
Dear Nathan,

Thanks for your answers, they really were helpful. 

1. Sorry for the first question, maybe I did not express myself properly. I was not expecting you to check term by term in my equation to check if it is right, I was more thinking about to have confirmation that the excitation moment around point A have to consider the terms surge, heave and pitch of the float reflected on point A. Anyways, I followed your suggestion, swapped the revolute A for a fixed constrained and got the moment at A. Compared the results from that simulation with the computed moment using the formula, and in principle looks the same, but there are some offset, any suggestion for these offset? I think can be explained by the fact that there is a constant moment at point A due to the vector difference between CoG and CoB.

[image]
[image]

2. Yes, I rerun the BEMIO. 
I check the structure of A matrix and corroborated with the number of states for each DoF. Nice explanation, now I think I can extract the information for each DoF.

One more thing related to this topic, I check the excitation impulse response given in hydroData, and found that there is 1 EIR for each DoF: 
[image]

But why is that for the radiation impulse response there is a 6x6 matrix (36 RIR) impulse responses?
[image]

Once again, thanks for your support,

Cheers

Juan Guerrero

nathanmtom, 2020-09-16:
@Liam-Guerrero Thanks for update on the progress for resolving issue. 

1. I believe you are correct that there will be an additional moment from the offset in the CoB and CoG. Furthermore, there is a static moment due to the weight of the attachment arm between the float and hinge A. The weight of the arm will also cause a moment about the hinge A. 
2. The reason that the RIR is a 6x6 matrix is because the hydrodynamic body has cross coupling terms between the 6 primary modes of motion. For example, when the body rotates in pitch (y-axis) the pressure imposed on the body from the rotation in the fluid may be unequal in the surge (x-direction) leading to a net force in the surge which results in a surge-pitch coupling term represented by the [1x5] term. Due to reciprocity relation the [1x5] term will be equal to the [5x1] term. Therefore, in calculating the hydrodynamic radiation forces in the surge (x-direction) theoretically requires the motion in all 6 degrees of freedom; however, several of the off diagonal terms can be zero especially if the body is symmetric.

I hope these answered have addressed your questions but as always feel free to ask follow on questions or comments to be illustrate the issues you are having. 

Cheers, 
Nathan
National Renewable Energy Laboratory

nathanmtom, 2020-09-16:
My apologies for accidentally closing this issue.

nathanmtom, 2020-09-26:
@Liam-Guerrero Please let us know if the last correspondence has satisfactorily answered your questions. We are happy to answer other questions you might have by submitting a new issue. 

Cheers, 
Nathan
National Renewable Energy Laboratory

Liam-Guerrero, 2020-09-27:
Dear Nathan, 

Yes, it really helps me to understand the issues I had.

Thanks for your help, very much appreciated.

Regards, 
Juan Guerrero

closed 2020-09-27 by Liam-Guerrero