WEC-Sim/WEC-Sim #412: waveClass('spectrumImport'): plotSpectrum

saltynexus, 2020-09-08:
This is a minor bug, but if you try to plot an imported spectrum using `plotSpectrum` in `waveClass` this line

`if obj.spectrumType == 'JS'`

will throw an error because `waves.spectrumType = 'spectrumImport'`. A simple fix would be

`if strcmp(obj.spectrumType,'JS') == 1`

kmruehl, 2020-09-17:
@saltynexus I'm unable to reproduce this error. I've run the RM3 example (attached) with the following wave settings:
```matlab
% Irregular Waves with imported spectrum
waves = waveClass('spectrumImport');        % Create the Wave Variable and Specify Type
waves.spectrumDataFile = 'spectrumData.mat';  %Name of User-Defined Spectrum File [:,2] = [f, Sf]
```
and was able to generate a plot of the imported wave spectrum using the `plotSpectrum` in the `userDefinedFunctions` post-processing script. 
``` matlab
waves.plotSpectrum();
```

[image]

saltynexus, 2020-09-18:
If you put a break point in the function, then evaluate
```
obj.spectrumType == 'JS'
```
you should get the following error.
```
Matrix dimensions must agree.
```

You don't need WEC-Sim to prove this is an error. Type in
```
'apple' == 'banana'
```

The reason why you get a plot is because the plot is generated before occurs and you're using
```
    try 
        wrapper.waves.plotSpectrum();
        saveas(gcf,[Dir.fig_param_set,sep,'Wave_Spectrum'],'png');close(gcf);
    catch
    end
```
Take out the `try` `catch` statements and you should be able to reproduce the error. It's hiding the fact an error exists.

kmruehl, 2020-09-21:
@saltynexus thanks for bringing this to our attention, the proposed solution has been added, and the warning should be removed.

commit c6292f90 (2020-09-21): resolved #412 warning in plotSpectrum
closed 2020-09-21 by kmruehl