WEC-Sim/WEC-Sim #413: waveClass('spectrumImport'): setWavePhase

saltynexus, 2020-09-08:
Matlab 2019a

I'm running an irregular case with a loaded spectrum. When I first start Matlab I get something along the following error:

> The current random number generator is the legacy generator. This is because you have executed a command such as rand('state',0), which activates MATLAB's legacy random number behavior. You may not use RNG to reseed the legacy random number generator. Use rng('default') to reinitialize the random number generator to its startup configuration, or call RNG using a specific generator type, such as rng(seed,'twister').

The error points inside `waveClass` function `setWavePhase(obj)` which sets the irregular wave phase. 

I'm not sure the exact source of the error, but I can only get past it if I add `twister` to `rng`. For example:

```
            if obj.phaseSeed ~= 0
                rng(obj.phaseSeed,'twister'); % Phase seed = 1,2,3,...,etc
            else
                rng('shuffle');         % Phase seed shuffled
            end
```

While that does get me moving, I'm not sure WEC-Sim is performing as initially intended following my ad-hoc fix. I think this should be looked at closer. [I found this addressed on Mathworks and it might help offer some insights.](https://www.mathworks.com/matlabcentral/answers/443801-random-number-generator-errors-after-switching-modes-from-state-to-shuffle).

dforbush2, 2020-09-10:
Hi @saltynexus 

Can you please confirm what matlab/wecsim version you are using?

dforbush2, 2020-09-14:
Thank you for confirming and proactively doing some research on this. 

I have a 2019a distribution installed locally and I am unable to replicate this error, I also can't seem to get it to arise in 2020a... is it possible you have something in your startup.m file, or anything else that you have set to execute at start-up that calls `rand` or `randn`? Quoting from the mathworks link you provide:

"Once "rand" or "randn" is used, "rng" cannot be used to reseed without reinitializing." so I'm wondering if this is the source of the error. 

I'd like to code in a try/catch for folks experiencing this error...but this is tough to do without being able to recreate it. Would it be possible to send me a *zip containing the model you're trying to run?

saltynexus, 2020-09-15:
I'll take a closer look later this week and try to give a better assessment.

dforbush2, 2020-09-16:
What OS are you running matlab on? That might be helpful.

saltynexus, 2020-09-18:
I'm having trouble reproducing this error now, so I'm not really sure what's going on. I'm going to close it for now and if I encounter it again in the future, I'll be sure to log things a little better.

closed 2020-09-18 by saltynexus