WEC-Sim/WEC-Sim #414: Scaling rm3 tutorial 1:100

aglassFHKiel, 2020-09-14:
Hi,

i have scaled the rm3 tutorial in 1:100 to compare the results with the measurements of an small scale rm3 laboratory model and got the error massage:

Error using wecSim (line 261)
['RM3/Global Reference Frame/Solver Configuration']: Error evaluating equations at time 0.05.  Stopping
simulation.  There may be a singularity in the solution.  If not, try reducing the step size (either by
reducing the fixed step size or by tightening the error tolerances)
Caused by:
    Error using wecSim (line 261)
    ['RM3/Floating (3DOF)/CONSTRAINT (Planar Joint)']: 'RM3/Floating (3DOF)/CONSTRAINT (Planar Joint)' has
    a degenerate mass distribution on its follower side.

[image]

For the geometry i used Soldiworks and Design Modeler.
Than i used AQWA R12020 for getting a ah1- and a lis-file to create the h5 file with bemio. 

[RM3_1_100.zip](https://github.com/WEC-Sim/WEC-Sim/files/5218672/RM3_1_100.zip)

Is it possible to scale in 1:100?

best regards
Andreas

aglassFHKiel, 2020-09-14:
In the picture you can sea that the sea level is very close to the sea bottom, thats because i changed the sea dimensions to the dimensions of our laboratory tank in aqwa.
x = 6m
y = 1.5m
z = 1m

[image]
[image]

Curious is that aqwa shows the 1:100 scale model and wecSim shows probably still the 1:1 model?
(i changed the stl files in the geometry folder)
Also curious is that the floater is on the right flotation depth and the plate not?

i have tried to solve the error but without an success.

can anybody help?

dforbush2, 2020-09-14:
The explorer visualization uses the *.stl files for the body geometry, which are distinct from those used in BEM that generate the *h5 file. So unless you have scaled this *stl file as well, that is the issue with the visualization. 

Additionally, looking at your mass properties (lines 65 and 71 in wecSimInputFile.m), it does not look like you have scaled your inertia properties. You're using the equilibrium flag to calculate the mass in heave based upon the displaced volume of the device. From the *h5 file you have provided, this has been scaled. Thus, you have a heave/surge mass that is many orders of magnitude different from your inertia. Finally, the PTO damping you have specified is still 1200000...this is again likely many orders of magnitude larger than you would want for your scaled device. The large differences between these orders of magnitude may be the cause of your solver instability. I scaled all of these parameters to be a common order of magnitude and my simulation ran for 5 seconds before generating the same error you saw. 

Finally, your *h5 file is a bit strange, as your excitation coefficients are a 6 x 6 matrix at each frequency, where WEC-Sim expects a 6 x 1 vector. Looking at your radiation coefficients as plotted by BEMIO, neither damping nor added mass approaches a horizontal asymptote at high frequencies, and actually seem to asymptote vertically. This suggests an issue in your BEM run and can certainly lead to simulation instability. I notice you're using a newer version of AQWA and your FORCERAO field of your AH1 file may be formatted differently. We have experienced a myriad of compatibility issues with AQWA due to their frequent changes to output file format and the variety of user-specified options that can affect this format, so it is possible that this is at the root of your issue.

In summary, I would suggest:
1). ensuring all inertial and PTO parameters are also scaled appropriately.
2). check your BEM result and processing approach.

aglassFHKiel, 2020-09-15:
Thanks for your suggestion. I going to check this.

Yes you are right with AQWA, i have to preparing always the lis-file manually before bemio runs.
Which output of externel program works better in bemio? WAMIT, AQWA or NEMOH?

aglassFHKiel, 2020-09-15:
Hi dforbush2,

thank you very much!

[image]

1. I changed the parameters in the wecSimInputFile.m line 65, 71 (inertia) and 81 (damping). I didn't know it that there are the parameter too.
2. The bemio plots doesn't converge until yet. AQWA is very sensitive with meshing. I don't know how i can improve the bemio plots? Also when i compare the 1:1 Model bemio file from WAMIT to AQWA. WAMIT looks very nice. AQWA looks bad. But the wecSim results look almost equal from WAMIT to AQWA.

dforbush2, 2020-09-21:
I can't give too many specific tips on BEM, but I do know that these codes are really sensitive to panel resolution. A 1:100 scale of RM3 is quite small, so individual panels in your mesh should be really small. A nice open source tool for checking and manipulating meshes is Meshmagick https://lheea.github.io/meshmagick/command_line/cli_usage.html#translations. I would try to
1). Ensure all of your meshes have a nice aspect ratio, so they are not overly long in one dimension so as to be short in another. 
2). Ensure you are using a sufficiently small mesh resolution. Often, this is an intentionally varied parameter, and mesh size will be reduced until results are shown to converge. 

Since WEC-Sim results depend on the BEM results, spending some time to get BEM you are confident in is worthwhile. Best of luck. I will close this issue but feel free to re-open or post another if you encounter an issue with WEC-Sim.

closed 2020-09-21 by dforbush2