WEC-Sim/WEC-Sim #415: Copy classes

saltynexus, 2020-09-15:
I'm running WEC-Sim using the MCR and I make some changes to the `simu` class during computation. For example, changing the time step to a smaller value does not invoke the default initialization behaviour with each iteration. I have to manually change all class variables to the appropriate values. I find it quite annoying that I can't make a copy of the class, then reassign the initial conditions utilizing WEC-Sim's default behaviour.

Here is what I'm trying to do (pseudo code)
```
test = simulationClass();
% input rest of settings
%
% lots of other code
%
% 
for j = 1:4
         test.dt = j/3;  % Change time step with each iteration
         wecSim;
end
```
This will break the next run, for many reasons, but one being the other time steps are not integer multiples.

I thought I could fix this utilizing WEC-Sim's default behaviour by making a copy of the class before running `wecSim` 
 ```
test = simulationClass();
% input rest of settings
%
% lots of other code
%
% 
test_copy = test; % Save all previous settings by making a copy
for j = 1:4 
         test = test_copy % Reinitialize with each iteration
         test.dt = j/3;
         wecSim;
end
```
In this case `wecSim` sees `test` as a new fresh input and invokes the default behaviour. This way I don't have to go through and manually change all values because `dt` changed. Unfortunately it is far from this easy! MATLAB creates pointers to the memory and therefore all changes in `test` will change `test_copy`.

Here is what I currently do to get the behaviour I want
 ```
test = simulationClass();
% input rest of settings
%
% lots of other code
%
% 
save('adhoc.mat','test'); % Save all previous settings by making a copy
for j = 1:4 
         load('adhoc.mat','test'); % Reinitialize with each iteration
         test.dt = j/3;
         wecSim;
         clear test
end
```
This works, but it's kinda "clunky" and it certainly doesn't seem efficient to be loading-clearing data with every iteration. I wanted to see if this could be improved. If I felt more comfortable with MATLAB's classes and understood how changes to the class would affect WEC-Sim as a whole, I would make the necessary changes and make a pull request. However, I think this one is above my skill set and I thought a developer might have better insights.

kmruehl, 2020-09-17:
@saltynexus I'm happy to help resolve this issue, but can you provide a little more context about what you're trying to accomplish? 

From what I understand, you're trying to run multiple cases of WEC-Sim, and for each case you would like to vary `simu.dt`. Is that correct? If so, the easiest way to do this is to set up a `wecSimMCR` case that varies `simu.dt`. I would recommend using [this MCR application](https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Multiple_Condition_Runs/RM3_MCROPT3), and modifying the `mcrExample.mat` file.  

If that is not the case, can you please clarify what you're trying to accomplish? And can you upload your model files?

saltynexus, 2020-09-18:
In a multi run, using MCR or not, you can NOT change `dt` without changing all other time steps manually because WEC-Sim wants integer multiples of `dt`. When WEC-Sim initializes a run, it dynamically sets all other required values using defaults. For example: 

RUN 1: I set `dt=1`, then WEC-Sim will automatically initialize the default nonlinear time step, output time step, etc. to 1 as well.

RUN 2: I set `dt=0.3`, WEC-Sim will crash because **previous** nonlinear time step, output time step, etc. are set to 1

One solution is to manually change all values that are affected by a change in `dt`. This is tedious and unnecessary. Another solution is to recreate the class, but I don't want to discard all my previous settings. I propose a solution to leverage WEC-Sim's initialization (like done in RUN 1 above) behaviour using a class template.

I'm trying to do the latter, where I only need to worry about `dt` (the parameter of primary interest). An easy way to accomplish this is to make a copy of the class before WEC-Sim initializes the default values in RUN 1. I can then use this class copy as a template, where the template is filled with each RUN (as outlined in my initial post). In this way, every RUN views the class a new one and thus initializes the class with default values (i.e. all time steps are correctly changed automatically).

The problem is I can not create a copy of WEC-Sim classes (simulation, wave, etc). Try this and you'll see what I mean

```
test = simuClass();
test.dt = 4;
test2 = test;
test2.dt = 0.32;
disp(test.dt)
```
I want `test2` to be an actual copy, not a pointer. In the example above `test.dt` changed even though that was not my intention. Therefore, the idea of building a template is not possible. 

I hope that is more clear.

kmruehl, 2020-09-21:
Instead of assigning the class to a new variable (and creating a pointer), I recommend saving the class, setting `simu.dt` to the desired value, and then saving/using the new class. This should resolve the pointer issue. 

```
clear all
load('RM3_matlabWorkspace.mat')
disp(simu.dt)
save('test1.mat','simu')

clear all
load test1.mat
simu.dt = 0.2;
save('test2.mat','simu')

clear all
load test1.mat
disp(simu.dt)
load test2.mat
disp(simu.dt)

```

saltynexus, 2020-09-25:
Yes, if you read the end of my initial post, this is what I currently do. However "it's kinda "clunky" and it certainly doesn't seem efficient to be loading-clearing data with every iteration". This was the reason why I asked about copying the class.

dforbush2, 2020-09-25:
Hi Salty,

I agree with @kmruehl, and I would second the use of this approach with a minor modification (no 'all' flag may a be a little faster):
```

clear
load('RM3_matlabWorkspace.mat')
disp(simu.dt)
save('test1.mat','simu')

clear
```
etc.

It sounds like it works for you? Even though this is "kinda clunky" the unwieldy aspects  of it are more of a MATLAB/Simulink problem than a particular shortcoming of WECSIM. The problem you point out with the time-step is particularly painful because of where this is evaluated in Simulink's internal initialization steps and because this setting is cached between simulink runs. Mathworks' intention I'd wager is to facilitate re-loading/restarting of a model... but this can be a problem when certain solver parameters need to change run-to-run. 

A potentially speedier approach may be to use the SimulationInput object to specify the parameters you wish to iterate, particularly note the PreSimFcn and PostSimFcn (See https://www.mathworks.com/help/simulink/slref/simulink.simulationinput-class.html and https://www.mathworks.com/help/simulink/slref/setmodelparameter.html). If this offers a significant speed-up and is satisfactorily un-clunky, let us know and we will see about using this method instead.

Have you tried running a variable-step solver (i.e. `simu.solver = 'ode45' `)? This may help alleviate the error message you are trying to work around, but I can't say for sure without seeing your model and input files.

saltynexus, 2020-09-26:
The save-load approach does work as intended, so yeah, it's not a problem/bug with WECSim. It was more of a question on whether or not there was a better implementation. From what I gather, MATLAB is more interested in conserving memory by creating a pointer as opposed to a deep copy...which is great, but not always desirable. They (MATLAB) could make it more transparent on how to create a deep copy, if so desired. I thought the WECSim developers might have more insight on how to go about it.

I'll check into the "SimulationInput" you brought to my attention and see if that offers any better solution. Again, this isn't an immediate issue.

dforbush2, 2020-10-01:
Thanks for keeping on this. 

For the purposes of other users: multiple simulations at different time-steps can be carried out using WEC-Sim MCR. The recommended example from @kmruehl https://github.com/WEC-Sim/WEC-Sim_Applications/tree/master/Multiple_Condition_Runs/RM3_MCROPT3 is a good starting place. From these source files:

1). create an MCR mat file (with variable name = mcr) in the style of mcrExample.mat. For example:

```
mcr = struct(); 
mcr.header{1} = 'simu.dt' % this must be a cell array
mcr.cases = [0.01; 0.05; 0.1]; % desired sim time steps, in seconds
save('myMCRFile','mcr'); % saves mcr file for future use
```

2) In the wecSimInputFile, point to this mcr file:

```
simu.mcrCaseFile = 'myMCRFile.mat'; % will load the mcr file made in step (1)
```

3). In userDefinedFunctionsMCR.m, log the variables of interest. For example, if we wanted to compare the resulting float heave position (body(1)) as calculated by each integration time step, the userDefinedFunctionsMCR.m could be:

```
nameField = strcat(['OutputLog' num2str(imcr)]);
mcr.(nameField) =  [output.bodies(1).position(:,3)];

```

This will create mcr.OutputLog1, mcr.OutputLog2, mcr.OutputLog3, which will contain data for the time steps specified in (1), in that order.

4). Run `wecSimMCR `from command line

As @saltynexus points out, it is likely this is sub-optimal in terms of speed, but it is functional, and at present is the recommended workflow.

closed 2020-10-01 by dforbush2

crobarcro, 2021-06-03:
> The save-load approach does work as intended, so yeah, it's not a problem/bug with WECSim. It was more of a question on whether or not there was a better implementation. From what I gather, MATLAB is more interested in conserving memory by creating a pointer as opposed to a deep copy...which is great, but not always desirable. They (MATLAB) could make it more transparent on how to create a deep copy, if so desired. I thought the WECSim developers might have more insight on how to go about it.
> 
> I'll check into the "SimulationInput" you brought to my attention and see if that offers any better solution. Again, this isn't an immediate issue. Feel free to close this issue if you need, or leave it open and I'll follow up in the future.

I'm not a WEC-Sim developer, just commenting

WEC-Sim uses [handle classes](https://uk.mathworks.com/help/matlab/handle-classes.html), matlab is very clear and [documents very well how to allow these classes to make deep copies](https://uk.mathworks.com/help/matlab/matlab_oop/custom-copy-behavior.html) but it requires work by the class developer. The alternative is to use value classes, a comparison of the two can be found [here](https://uk.mathworks.com/help/matlab/matlab_oop/comparing-handle-and-value-classes.html)

I'm sure the WEC-Sim devs would welcome a pull request that implemented a copy method for their classes to allow deep copies to be made!