WEC-Sim/WEC-Sim #417: How can I change the XLables in the BEMIO Plots

aglassFHKiel, 2020-09-23:
[image]

How can i change the time in normalized radiation and normalized excitation irfs?

aglassFHKiel, 2020-09-23:
up to 60 secounds

dforbush2, 2020-09-24:
After BEMIO runs, click the plot you're interested in and type

`xlim([0 60]);`

However, the time over which the IRF is calculated is a user input to `Radiation_IRF` and `Excitation_IRF`, and the X-limit of the plots created should be inherited from this input.

For example, 60 second IRFs are calculated as:
```

hydro = Radiation_IRF(hydro,60,[],[],[],[]);
hydro = Excitation_IRF(hydro,60,[],[],[],[]);
```
Please see https://wec-sim.github.io/WEC-Sim/advanced_features.html#bemio

aglassFHKiel, 2020-09-24:
perfekt, thanks

closed 2020-09-24 by aglassFHKiel