WEC-Sim/WEC-Sim #418: about the multi-directional wave analysis, the h5 file generated by the AQWA output.

abaddonwang, 2020-09-24:
When I used AQWA to get the .AH1 file and .LIS file, I can select how many wave directions and wave frequencies I'd like to use. In AQWA, I used 99 frequencies and the default 8 wave directions.

I got the .h5 file for 2D waves successfully, I think it's enough for single-direction wave analysis since the wave anlge is 0 degree. However, when I tried to carry on the multi-directional wave analysis in WEC-SIM. I'm not sure how BEMIO calculate the .h5 file based on the .AH1 and .LIS files.

Do I need to recalculate the .h5 file with more wave directions (like 360 directions)?

ThBoerner, 2020-09-24:
Hi @abaddonwang ,

By default, BEMIO will check your .AH1 and .LIS files obtained from Aqwa for the number of wave headings and include all of them in the .h5 file. This happens in the "Read_AQWA.m" file:
line 33 reads the AH1 file
line 54 writes the relevant information into a temporary variable, and 
line 66 extracts the number of wave headings that information is available for.

The same script then loops through each of the wave headings and writes hydrodynamic properties and excitation forces for each wave heading into a matlab matrix that is later on saved as the .h5 file.

There are multiple ways how to check if BEMIO correctly writes information for all desired wave headings into the .h5 file. The easiest ones would probably be the following:
1. You can actually open .h5 files with an .h5 file viewer (e.g. HDF View which is free) and browse through the file structure and content. In there you should be able to find an entry that lists all available wave headings.
2. After you ran BEMIO to convert your .AH1 and .LIS files into the .h5 file you will have a MATLAB structure called "hydro" in your workspace. Have a look inside and search for the entry "hydro.Nh" which should list all wave headings that were processed by the BEMIO routine.

----

If you .h5 file includes all your wave headings, you can use the "waves.waveDir" command in wecsim to define your incident wave direction. Note, that when you use a wave direction in between two available wave directions, wecsim will interpolate the hydrodynamics information available in your .h5 file. This allows you to run wave directions that you did not explicitly calculate in your BEM solver, aka Aqwa. Depending on your asymmetry of your body geometry this will be more or less accurate so make sure to potentially do a sensitivity analysis on your wave heading.

Hope this helps.

kmruehl, 2020-09-30:
@ThBoerner Thank you!

@abaddonwang did this resolve your issue?

abaddonwang, 2020-10-01:
Thank you very much, I will check the .h5 file and BEMIO.

Best regards,

Abaddon

[http://www.plymouth.ac.uk/images/email_footer.gif]<http://www.plymouth.ac.uk/worldclass>

This email and any files with it are confidential and intended solely for the use of the recipient to whom it is addressed. If you are not the intended recipient then copying, distribution or other use of the information contained is strictly prohibited and you should not rely on it. If you have received this email in error please let the sender know immediately and delete it from your system(s). Internet emails are not necessarily secure. While we take every care, University of Plymouth accepts no responsibility for viruses and it is your responsibility to scan emails and their attachments. University of Plymouth does not accept responsibility for any changes made after it was sent. Nothing in this email or its attachments constitutes an order for goods or services unless accompanied by an official order form.

closed 2020-10-12 by dforbush2