WEC-Sim/WEC-Sim #419: Simulation stuck in the pre-processing stage, when small time-step, such as 0.0001s

zhykly, 2020-09-29:
Hi someone,

to hybrid with other simulation software, I need a very small time-step.

Take RM3 for example, when I set the time-step as 0.0001s.  Then I start to run wec-sim( Only wec-sim, not hybrid). 
After a few hours, WEC-sim is still in the pre-processing stage. and it seems being blocked.

[image]

wave information is:
% Regular Waves  
waves = waveClass('regular');           % Initialize Wave Class and Specify Type                                 
waves.H = 2.5;                          % Wave Height [m]
waves.T = 8;                            % Wave Period [s]

All other paramether take same as the standard examples case, RM3.

is there any methods to reduce the pre-processing calculation burden? 
how to resolve the issue?

Win 10+ Matlab 2018a
Intel Core i7-9700+ 16GB RAM

millions thanks....

best regards
York

dforbush2, 2020-09-30:
Hi York,

I tried this, and it appeared that most of the time was taken up with the waves.waveSetup function, in which the wave time-series is calculated.  

If you're using a time step of 0.0001 s. with the simulation time of 400 s, this pre-calculation is generating numerous arrays with a length of 4e6. It is probable that this will take quite awhile, though you shouldn't be running up against your RAM limits.

May I ask why you are simulating at such a small time step? Typically, 1/00 of the dominant wave period or so will be sufficient. If there are potentially some fast dynamics, such power electronics, to be considered, I would recommend you change to a variable-step or stiff solver.

For example:
`simu.solver = 'ode45' `

This will allow you to specify a larger simu.dt in the pre-calculation, greatly speeding these calculations up. By default, the variable-step solver will never exceed your specified simu.dt, but in theory, the adaptive time-step of a variable step solver should still allow you to accurately capture fast dynamics, should they prove significant.

yuyihsiang, 2020-09-30:
Few things I would like to add:

- Use rapid-accelerator mode: 
    simu.mode = 'rapid-accelerator';

- Use the state-space calculation option: 
    simu.ssCalc = 1;

- Shorten the Simu.CITime: Note please make sure the time is sufficient to have the impulse response function converged

- You may be able to use two solvers, one for the basic WEC-Sim simulation and the other one for the other simulation software, which allows you to have different time step sizes for different problems. May I ask what is the other simulation software?

--Yi-Hsiang

zhykly, 2020-10-01:
> Hi York,
> 
> I tried this, and it appeared that most of the time was taken up with the waves.waveSetup function, in which the wave time-series is calculated.
> 
> If you're using a time step of 0.0001 s. with the simulation time of 400 s, this pre-calculation is generating numerous arrays with a length of 4e6. It is probable that this will take quite awhile, though you shouldn't be running up against your RAM limits.
> 
> May I ask why you are simulating at such a small time step? Typically, 1/00 of the dominant wave period or so will be sufficient. If there are potentially some fast dynamics, such power electronics, to be considered, I would recommend you change to a variable-step or stiff solver.
> 
> For example:
> `simu.solver = 'ode45' `
> 
> This will allow you to specify a larger simu.dt in the pre-calculation, greatly speeding these calculations up. By default, the variable-step solver will never exceed your specified simu.dt, but in theory, the adaptive time-step of a variable step solver should still allow you to accurately capture fast dynamics in simulation, should they prove significant.

Professional answers! thanks.
Yes, indeed, I am doing a simulation involving power electronics and high stiffness spring. This makes the differential equation have a very small time constant.
For the differential equations with very small time constant. Variable step size algorithm is indeed an effective way to deal with it.

However, in some scenarios, the fixed step size algorithm has an indispensable role.

I checked the waves.waveElevReg function code, which is called by waves.waveSetup function. and I found that _waves.waveAmpTime1, waves.waveAmpTime2 and waves.waveAmpTime3_, did not pre-allocate memory during the loop. It seems be the reason for slowing down speed.
https://github.com/WEC-Sim/WEC-Sim/blob/c6292f9045c4be70753fd2060431d775993cb15e/source/objects/waveClass.m#L568

Besides, for most simulations, I guess “Wave Gauge” is not necessary. Can we consider not to use “Wave Gauge” by default?
It will greatly reduce the simulation cost in pre-processing.

best wishes
York

zhykly, 2020-10-01:
Potentially, there are two software that might be used. it is still in the stage of technical route trial.
a)	Power electronics, including Rectifier (AC-DC) device and Voltage stabilizing device (DC-DC) device.
b)	ANSYS fluent, to calculate the viscous drag of underwater body. The time step is estimated to be 0.005 s.

dforbush2, 2020-10-01:
Hi York,

Great comment on the Wave Gauge amplitudes being calculated unnecessarily. In reviewing this issue yesterday that also came to my attention, and has been added for now as a project for the team. This should reduce pre-calculation time by up to 4x, and will be addressed in the near future. And you are also correct that at the very least WaveAmpTime[1:3] should be pre-allocated. I will push this pre-allocation modification right now. 

I don't have particular suggestions for coupling with ANSYS. Could you provide a little more detail with respect to this coupling? If the two are not running simultaneously, you may be able to run Simulink with a variable time-step, but develop outputs at specified fixed interval. https://www.mathworks.com/help/simulink/ug/specifying-which-samples-to-export-for-models-with-variable-step-solvers.html (see "Produce Additional Output").

Simulink does have some Power Electronics modeling options, contained in Simscape Electrical (an additional package). This would offer a seamless way to include these components in a model, and might offer a nice intermediate step.

zhykly, 2020-10-02:
Hi dforbush2,

Thanks for your help, it definitely helps me. and I will keep on trying the technical route. 

Anyway, thank you again, and I wish WEC-sim would become more and more perfect and powerful.

Best wishes,
York

closed 2020-10-05 by dforbush2