WEC-Sim/WEC-Sim #420: Small error

saltynexus, 2020-09-29:
Undefined function or variable 'value'.

Error in simulationClass/checkinputs (line 154)
                error('The simMechanics file, %s, does not exist in the case directory',value)

I think `value` is intended to be the name of the file

dforbush2, 2020-09-30:
Hi Salty,

Can you provide a bit more context about when and how this error appeared? Its not one I have encountered in my regular use of WEC-Sim

saltynexus, 2020-09-30:
I pointed out exactly where the error is. I don't know what more I can say. If you want to reproduce it, don't include the `simMechanics` file in the case directory. The actual error is "Undefined function or variable 'value'." not the fact that the file is missing.

dforbush2, 2020-09-30:
Oh I understand. It's a problem with producing a useful error message. Not an error you were encountering and unsure how to get around. 

I'll see about fixing this.

saltynexus, 2020-09-30:
Yeah...it is a small error I came across and was sharing....nothing of major concern. Overall, I'm trying to keep the developers informed of things I see while working with WEC-Sim. Would you prefer that I don't open issues of this kind?

dforbush2, 2020-09-30:
Please do, its definitely helpful and appreciated. I just initially misunderstood.

closed 2020-10-01 by dforbush2