WEC-Sim/WEC-Sim #422: PTO k, c, and linearDamping values?

sugarl14, 2020-10-07:
Hi,

I was wondering what you all did to determine the values for k, c, and linearDamping for the OSWEC following tutorials:

1) for the oswec wecSiminputfile:
pto(1).k = ?
pto(1).c = ?

and 
2) for the PTO-Sim example for the OSWEC
body(1).linearDamping(5,5)= ?

I also notices for the PTO-Sim example (OSWEC_Hydraulic_PTO), you all defined pto(1).k and pto(1).c to be zero. Why did you choose this?

My OWSC geometry is a little different in size so I need to calculate my own values for k, c, and linearDamping!

Thank you!

dforbush2, 2020-10-07:
Hi @sugarl14 

With respect to 1), the tutorial instructions are given on https://wec-sim.github.io/WEC-Sim/tutorials.html#oscillating-surge-wec-oswec . It appears that for this example that pto(1).k = 0 and pto(1).c = 0. The intention was to provide a sandbox for the user to experiment with these values. Note that even with no stiffness and damping added at the pto, there is still stiffness and damping in the model due to hydrostatic restoring stiffness and radiation damping. 

Considering (2), the equivalents of pto(1).k and pto(1).c are functions of the parameters specified in the ptoSimInputFile.m and the PTO system selected. Additional linear damping (e.g., body(1).linearDamping(5,5)) captures damping sources external to the PTO, such as bearing friction or fluid drag.

I'm not sure that about the specifics of any calculation that informed the selection of these values for the example cases. However, with regard to how to specify these parameters, I would recommend you consider the pto stiffness and damping as controller gains. With this, you can specify the damping and stiffness based upon the real and imaginary parts of the device impedance (estimated from BEM hydrodynamic coefficients), a method outlined in Cho et. al (2019) https://www.researchgate.net/publication/338374910_Evaluation_of_predictionless_control_for_wave_energy_converters, specifically equations 3, 8, and 9.  

With respect to linear damping specification, it depends on the physical source of the friction you are trying to model. If it is a mechanical friction, like bearing drag, that is likely captured by a linear equation F_{friction} = -c * velocity, then I would attempt to select a value reasonable for c from the physical hardware you are trying to represent. Some examples: https://evolution.skf.com/us/damping-in-a-rolling-bearing/. 

If, however, the damping source is more likely viscous fluid drag, obeying the quadratic relation F_{friction} = -0.5 * Area * Cd abs(Vel) * Vel, then I would recommend these values be based upon the geometry and projected area of your moving body, some examples: https://en.wikipedia.org/wiki/Drag_coefficient .

sugarl14, 2020-10-08:
Thank you for the information!

dforbush2, 2020-10-14:
Yes. If the user specifies hydraulic PTO parameters in the ptoSimInputFile using PTO-Sim and is using the appropriate blocks from the PTOSim_Lib.slx in the simulink model, then the parameters specified in the ptoSimClass object will be used in the simulation. (For reference: https://wec-sim.github.io/WEC-Sim/advanced_features.html#pto-sim)

The pto(1).k and/or pto(1).c parameters are used by the pto blocks in the WECSim_Lib blockset. 

Generally, the blocks in the simulink model read the parameters in the <blank>Class objects specified in the wecSimInputFile.

sugarl14, 2020-10-15:
Awesome. Thank you very much!

closed 2020-10-15 by sugarl14