WEC-Sim/WEC-Sim #424: .AH1 structure ordering

sugarl14, 2020-10-08:
Hi,

I am using AQWA to model my own OWSC and just realized something that may or may not be causing some issues when I run wec-sim. In my specific case, I have 2 parts-- a base and a flap. And I noticed in my .AH1 file that the 1st structure being analyzed is the base, and the 2nd structure is the flap. However, I feel like it should be the other way around since in Simulink, body(1) is the flap, and body(2) is the base. Also, in the wec-sim folder examples>BEMIO>AQWA> *open .AH1 file* , I believe the flap is first, followed by the base. 

So my question is this: Does the ordering of structures in the .AH1 file affect the final results in WEC-Sim? I've attached a snipped of my issue if I didn't explain it clear enough above. 

[image]

dforbush2, 2020-10-08:
Hi @sugarl14 

If you are modeling both the flap and the base as hydrodynamic bodies, then order does not matter. If, however, as in some WEC-Sim examples, you are modeling the flap as a non-hydrodynamic body, you may have to list the hydrodynamic body first (i.e., body(1)), which can create issues if your BEM output has this listed as body 2. 

Let me merge pull request #423. This was intended to add an unrelated feature, but incidentally should fix this ordering limitation as well. I recommend your body numbers as specified in your WEC-Sim input file align with those in the BEM code.

dforbush2, 2020-10-14:
@sugarl14 Did this answer your question?

sugarl14, 2020-10-15:
Yes! Thank you!

closed 2020-10-15 by sugarl14