WEC-Sim/WEC-Sim #427: OWSC Rotation Limit

sugarl14, 2020-10-14:
Hi,

This issue is similar to issue #343  , but it is dealing with an OWSC.  Whenever I run a simulation on my OWSC geometry, the flap seems to rotate beyond the base structure (which, physically, is impossible). I have attached a picture of this below. Is it possible to model end stops in WEC-Sim so the flap will only rotate say + and - 45 degrees?

[image]

[image]

[image]

dforbush2, 2020-10-14:
Hi @sugarl14 

I think you should be able to do something similar to the solution proposed for #343 by specifying a user-prescribed force on your rotational PTO, you could also try the hard-stop Simscape blocks (external to the WEC-Sim distribution) suggested in #185.  

Additionally, later versions of MATLAB/Simulink have additional dialog options on constraint/PTO type blocks that allow the user to specify end stops a manner mathematically identical to that suggested in #343 but directly in the block dialog. For me in 2020a, for a translational PTO, you can access these parameters from the "Limits" drop-down, in the screen shot:

[image]

However, these options are NOT available in older MATLAB versions (not sure exactly which, but roughly 2018a).

sugarl14, 2020-10-15:
Great. Thank you for all the help!

closed 2020-10-15 by sugarl14