WEC-Sim/WEC-Sim #429: source code mangement

kmruehl, 2020-10-15:
Opening a issue to tag a thread on source code management. This is an ongoing discussion to determine the best workflow for collaborative development on WEC-Sim. 
https://github.com/SNL-WaterPower/WecOptTool/issues/192

H0R5E, 2020-10-16:
One important practice I would recommend when merging PRs is to use the [Squash and Merge](https://docs.github.com/en/free-pro-team@latest/github/collaborating-with-issues-and-pull-requests/about-pull-request-merges#squash-and-merge-your-pull-request-commits) option, which will roll all of the file changes in a PR into a single commit. The main reason this is important is that if there was a mistaken upload of a large file on the PR's feature branch, which was then subsequently deleted, then the history of this file addition is not added to the main repository. Secondly, it provides an opportunity to summarize the changes in the PR in a single commit message, removing any inconsequential commits (e.g. typo fixes).

kmruehl, 2020-10-26:
@H0R5E I/we agree with the squash and merge option. And I agree with your comment about repo size. For now we will be working off of forks and submitting PRs into the dev branch. But we may need to re-evaluate the best workflow for features that need development by more than one person.

crobarcro, 2020-12-14:
I have a suggestion, all the code which is currently inside simulink a function blocks should be made into standalone m files, and then the function blocks should just call these files with a thin wrapper. The advantage will be that normal code tools like github can then track changes to the file and show diffs etc.

Currently figuring out what changes in the hydrodynamics code in a commit is a real pain. Simulink files just look like binary blobs to git. Not very helpful for auditing changes to the code.

kmruehl, 2020-12-18:
@crobarcro that's a good suggestion. We will look into how we can better manage the WEC-Sim Library, specifically how we call scripts/functions within the Simulink file.