WEC-Sim/WEC-Sim #431: A question about PTO-Sim

qorbanpour, 2020-10-21:
I am new to WEC-Sim and I was reading the tutorial about PTO-Sim  :  [](https://wec-sim.github.io/WEC-Sim/advanced_features.html#pto-sim)  . 

In the tutorial it is mentioned as follow : 

 **In summary, the files need to run RM3 with PTO-Sim case are the following:

WEC-Sim input file: wecSimInputFile.m (make sure to set the PTO linear damping to zero)**

My question is :

Why the damping should be set to zero ( my guess is that it's calculated in PTO-Sim block somewhere ) . Also Is the stiffness from PTO block needs to be set to zero ! 

Or better putting in this way,  Having a PTO-Sim block , what is the role of stiffness and damping from PTO-block , considering that all PTO-Sim blocks that comes with WEC-Sim example are hydraulic or  Direct Drive PTO  !  

Warmest Regards,
Sahand.

dforbush2, 2020-10-26:
Hi @qorbanpour 

"Why the damping should be set to zero (my guess is that it's calculated in PTO-Sim block somewhere )"

Your guess is correct! When using PTO-Sim blocks, these parameters (k and c) are functions of the specific PTO component parameters (e.g., motor constants and friction, etc). The k or c specified in the simple ptoClass blocks (pto.k and pto.c) are not used by the PTO-Sim blocks, and are external to them. You could (I suppose) use the pto.k and pto.c to account for some sources of stiffness external to the system described in PTO-Sim, but in the example you are discussing, there were no external stiffness and damping terms, and so pto.k = 0 and pto.c = 0.  

Be aware this extends to calculated quantities like power: the power calculated in the ptoClass block will, in this example, be zero, but the power in the hydraulic or direct-drive PTO, which represents the power used/absorbed by the device, is non-zero.  

I understand how this can be confusing: we are revising our documentation to better address this.

Does this answer your question?

Best,
Dom

jleonqu, 2020-10-27:
Hi @qorbanpour 

In the example that you mentioned the hydraulic PTO is used. The hydraulic PTO is receiving an input from the Translational PTO Actuation Force. If you take a look inside this block, you can see that there is a subsystem called InternalMechanics.

[image]

This subsystem is versatile and you can use it depending on your specific needs. As default, the subsystem is solving the dynamic equation for a translational PTO mechanism represented as a linear spring-damper system, which is:

F_pto = -K_pto*x_rel - C_pto*v_rel

This equation can change depending on the characteristics of the systems that you want to simulate. This can be used to simulate, for example, the energy loss by friction in the mechanical connection of the PTO or it can be used to simulate your own model for a new PTO system. In the specific case of the example with the hydraulic PTO, the damping and the stiffness are zero because otherwise it would be like if you were using two PTO systems at the same time.

I hope this can help you with your simulation.

Best,

Jorge Leon

kmruehl, 2020-10-27:
@qorbanpour Does this answer your question?

qorbanpour, 2020-10-27:
@dforbush2  Thank you for the explanation . 
@jleonqu  Thank you for the nice explanation .
@kmruehl  Yes, it helped alot . Thanks for clarification and your helps .

Warmest,
Sahand.

kmruehl, 2020-10-27:
Of course, I'm going to close this issue.

closed 2020-10-27 by kmruehl