WEC-Sim/WEC-Sim #432: WEC-Sim & FAST

enrjcutl, 2020-10-21:
Is there a reference to coupling the two codes WEC-Sim and FAST?

yuyihsiang, 2020-10-26:
I don't think there is a direct way to couple the two codes. May I ask what is the objective of the effort?

--YI-Hsiang

enrjcutl, 2020-10-26:
Thanks for your response Yi.

I came across an extract by NREL stating WEC-Sim can be integrated with other tools. https://www.nrel.gov/news/features/2020/wave-energy-converter-simulator.html, reinforced by a paper by Li et al. found here https://doi.org/10.1016/j.renene.2017.09.080. 

The scope of my research is very similar. I am studying the performance of a integrated floating energy system that combines a floating wind turbine, multiple WECs and tidal turbines. I essentially need to develop my own coupling framework between WEC-Sim and FAST in order to evaluate the global system performance. 

Kind Regards

Josh Cutler

ttran18, 2020-10-26:
Hey Josh,

FYI:  There is a group at Swansea University, the UK is also being developed the floating energy system (wind+tidal) using WEC-Sim code. The main author presented this research in the PRIMaRE 7th Conference. His name is Jack Hughes.

Cheers,
Toan

enrjcutl, 2020-10-28:
Hi Toan,

I will look further into this. 

Many thanks,

Josh

kmruehl, 2020-10-28:
@ttran18 and @yuyihsiang thanks! @enrjcutl I'm going to close this issue.

crobarcro, 2020-12-14:
You might be interested in our fork of WEC-Sim, which uses [MBDyn ](https://www.mbdyn.org/)as the multibody dynamics engine instead of Simulink. The advantage is that MBDyn is already used to simulate wind turbines, and in fact previously had a plugin to use aerodyn from FAST inside a simulation as the input to the turbine (but this would need updating, last working version was aerodyn 12). You can use any MBDyn elements inside a simulation. 

Our fork is part of RenewNet Foundry [here](https://sourceforge.net/projects/rnfoundry/). Documentation on our fork is [here](https://rnfoundry.sourceforge.io/).

Our version is also Octave compatible (a free clone of Matlab). 

Our fork also has potential to [correctly implement added mass](https://public.gitlab.polimi.it/DAER/mbdyn/-/issues/58) which is restricted in simulink.

kmruehl, 2020-12-18:
@crobarcro thank you for responding to this issue and sharing the fork with MBDyn. The @WEC-Sim/wec-sim-lab-team would like to revisit inclusion of MBDyn with WEC-Sim. What would be the best way to discuss? Thank you!

Relates to #160

crobarcro, 2020-12-19:
Happy to have a discussion about this, I'd be interested in merging my fork and wec-sim. Actually they could currently coexist, I use namespaces in matlab so all my classes are referenced like:
```
wsim.body
wsim.waveSettings
```
I can see a range of ways the two projects could coexist or merge.

From my end there isn't really the resources to keep the hydrodynamics up to date with WEC-Sim as much as I would like, so if we could get support from the official WEC-Sim team to do this in some capacity it would be fantastic. There may even be ways to merge the two codebases, that don't significantly increase the maintenance burden.

I have also been slowly moving some functionality from being implemented in Matlab to being implemented in MBDyn. 

Anyway, I have attempted to connect on linkedIn to discuss this further privately.

enrjcutl, 2021-01-20:
@crobarcro I apologise in advance for the late response. Regarding your post on 14/12/2020 it is definitely something I am interested in. My objective is to simulate a hybrid floating energy system (wind+wave+tidal). Would it be possible to have a conversation about this?

kmruehl, 2021-02-02:
@crobarcro I finally logged into LinkedIn and responded to your message. Looking forward to discussions about how we can collaborate. 

@enrjcutl I'm going to go ahead and close this issue because it's a bit beyond the current scope of WEC-Sim and it looks like @crobarcro already provided some great resources.

closed 2021-02-02 by kmruehl