WEC-Sim/WEC-Sim #436: Moving docs into master and automatic deployment

H0R5E, 2020-10-27:
This issue is to demonstrate the concept of storing documentation in `master` (and any branches off master), in the `docs` folder, and using TravisCI to build and deploy the docs to the `gh-pages` branch. I have set up an example of this process in [this repository](https://github.com/H0R5E/WEC-Sim-Test). The associated docs (built by [TravisCI](https://travis-ci.org/github/H0R5E/WEC-Sim-Test/builds/738348469)) can be found [here](https://h0r5e.github.io/WEC-Sim-Test/).

There are a few extra goodies here also, as follows:

1. By using [sphinxext-remoteliteralinclude](https://github.com/wpilibsuite/sphinxext-remoteliteralinclude) I was able to remove the need for the WEC-Sim Applications repo to be cloned for the docs to compile. The example code is now read from Github.com, directly.
2. I have used [sphinxcontrib-versioning](https://github.com/sphinx-contrib/sphinxcontrib-versioning) to build the docs for both the `master` and the `dev` branches. This allows docs updates to be submitted against dev which can act as a "staging" area, before merging into master.
3. I have added a folder structure to the `.rst` files, so that they are kept in their own folder called `man`, which makes the sphinx system a little easier to navigate.

@kmruehl, if this is something you would like to implement, then can I request that we stop adding to the `gh-pages` branch. I have to manually merge changes between the two systems to keep them up to date.

kmruehl, 2020-10-27:
@H0R5E definitely. Thank you so much for getting this set up. It looks great! We will pause all documentation updates until this PR is submitted and merge. Do you want to submit this PR to both master and dev, or just dev? FYI, it's our goal to merge dev into master by the end of this week. 

@WEC-Sim/wec-sim-lab-team do **not** update the docs until @H0R5E has submitted a PR for this feature and we've merged it..

H0R5E, 2020-10-27:
@kmruehl, the easiest was is probably if let me have full control of the repository and the travisCI account for a short while so I can get this done. It often needs a little bit of tweaking to get it to build correctly. I don't have time to do it today, but if you give me permission to do so, I can attempt it tomorrow morning, my time.

One thing which you might want to do first is make sure that TravisCI is set up to use [travis-ci.com](https://travis-ci.com/). Builds have become super-slow on travis-ci.org, because they are [migrating all the open-source accounts](https://docs.travis-ci.com/user/migrate/open-source-repository-migration). It definitely seems worth doing now.

kmruehl, 2020-10-27:
I'll set this set up for you right now.

H0R5E, 2020-10-28:
@kmruehl I think that the second submodule for WEC-Sim_Applications (in the `applications` folder) is not required any more. Do you want me to remove it as the last task in this issue?

kmruehl, 2020-10-28:
@H0R5E we actually no longer need either of the submodules so I deleted them both on dev. Please take a look and see if I missed anything.

H0R5E, 2020-10-29:
Hi @kmruehl, I didn't think you would want to remove both, because one of them is explicitly mentioned in the docs:

http://wec-sim.github.io/WEC-Sim/man/tutorials.html?highlight=submodule#applications-repository

kmruehl, 2020-10-29:
@H0R5E Thanks for bringing my attention to that note. It's outdated and I'll revise it.

closed 2020-10-29 by kmruehl