WEC-Sim/WEC-Sim #437: Bug when using B2B and CI calculation together

akeeste, 2020-10-27:
When running the WEC-Sim_Applications repo body-to-body case 4 (B2B and CI calculation). A dimension error is generated in the convolution integral calculation within the radiation force calculation block.

the full error message is: 
Error using wecSim (line 292)
Non-singleton dimensions of the two input arrays must match each other.
	Error in 'RM3/Float/Hydrodynamic Body/Wave Radiation Forces Calculation/SS CI and Constant-Damping-CoeVariant
 Subsystem/Convolution Integral Calculation/MATLAB Function1'.

Within that function, line 21 throws an error because IRKB_interp and velocity are different dimensions:
time_series = bsxfun(@times, IRKB_interp, velocity);

dforbush2, 2020-10-27:
Hi @akeeste ,

I can't replicate this error. Is it possible this is path issue on your local machine? Perhaps it is pointing to the wrong BEM data?

kmruehl, 2020-10-28:
@akeeste and @dforbush2 I'm also not able to reproduce this issue. I'm currently running MATLAB 2020a and I've pulled the latest version of WEC-Sim_Applications https://github.com/WEC-Sim/WEC-Sim_Applications/commit/34c3ad5bfce8e3ac296f686683df9c9457324185, and the latest commits to both the dev c6e768053157f7a6dfea92fb6781557e6b14144f and master ac8af2a12f7212cb5c813b3e92e0e2d0bfe56c6a branches of WEC-Sim. I'm able to run all of the B2B cases with both dev and master.

[image]

akeeste, 2020-10-28:
This issue appears to only appear in MATLAB R2020b. 2020a is fully functional. I believe the problem is coming from a change (or bug) in how the bsxfun() function is implicitly expanding the velocity and IRKB_interp arrays.

Up to the point of the error, both versions contain the following dimensions (at least when running the B2B/case4 application).
V - [6]
IRKB - [601x6x12]
lenJ - [1]
CTTime - [601]
velocity - [6 601]
IRKB_interp - [12 601 6]

2020a results in the following variable dimensions:
time_series - [12 601 6]
sum(times_series,1) - [1 601 6]
trapz('') - [1 1 6]
F_FM - [6 1]

When times_series is calculated in line 21, the function sees the following dimensions being multiplied together:
[12 601 6] .* [6 601] -> [12 601 6]

It seems like 2020a expands this to work correctly, but 2020b does not. I copied the ConvolutionIntegral_interp function below for convenience:

``
function F_FM = ConvolutionIntegral_interp(v, IRKB, lenJ, CTTime)
%#codegen
% Function to calculate convolution integral

%define persistent variables
interp_factor = 1;
persistent velocity IRKB_interp CTTime_interp;

if isempty(velocity) 
    CTTime_interp  = CTTime(1:interp_factor:end);
    velocity       = zeros(lenJ,length(CTTime_interp));
    IRKB_reordered = permute(IRKB,[3 1 2]);                     % permutes from [601 6 12] to [12 601 6]
    IRKB_interp    = IRKB_reordered(:,1:interp_factor:end, :);  % what does this do?
end 
% shift velocity
velocity      = circshift(velocity, 1, 2);
v = v(:)'';           %force v to be a column
velocity(:,1) = v(:);

%integrate
time_series = bsxfun(@times, IRKB_interp, velocity);
F_FM = squeeze(trapz(CTTime_interp,sum(time_series, 1)));

end
``

dforbush2, 2020-10-28:
Sounds like some permute() or reshape() to explicitly set array size might be a more robust solution that relying on bsxfun to sort out the proper dimensions? I'm guessing that 2020b documentation for bsxfun might mention the nature of the change more explicitly?

akeeste, 2020-10-28:
@dforbush2 Yes I agree permuting them so that the dimensions match exactly before the multiplication is more robust going forward. I'll submit a PR once I get it working

PR #444 (2020-12-02): Fix for CI-B2B bug
This is a fix and cleanup of the convolution integral function. There is a version bug between MATLAB R2020a and R2020b in the bsxfun() function.

This closes #437 .

Note it does not pass all tests, 8 fail. 6 test have minor differences close to the tolerance limit, and two in the passive yaw irregular case fail with larger errors.
Pos_diff = 2.5e-4
Force_diff = 7.9e+3
closed 2020-12-02 by akeeste