WEC-Sim/WEC-Sim #447: Information about wecSim Outputs

hossein-mehdipour, 2020-10-30:
I wanted to ask if there is any information about wecSim outputs, that explains what are they exactly, for example in  output.bodies(1).position  &  output.bodies(1).velocity   I'm not really sure what each column represents or what is its dimension, (for example, does column #5 represent the velocity in pitch, and its in radian)?
Or what's the difference between output.ptos.forceInternalMechanics, output.ptos.powerInternalMechanics  & output.ptos.forceTotal?
Can I find this information anywhere? (in like a manual or an article or somewhere else)

And also thank you so much for putting the time to answer every single issue the community has, it's highly appreciated.

kmruehl, 2020-11-19:
@hossein-mehdipour Thank you for your question, and sorry for the delayed response. The only place we document the output structure is here: http://wec-sim.github.io/WEC-Sim/master/man/api.html#response. However, I did develop a more detailed description of the WEC-Sim outputs here: https://mhkit-software.github.io/MHKiT/mhkit-matlab/wecsim_example.html. Based on this request, I plan to update the documentation to include more information about WEC-Sim outputs, and how to use them.

hossein-mehdipour, 2020-11-20:
@kmruehl Thank you for the more detailed description page, it will give better insight about WEC-Sim outputs for new users for sure, and I'm happy that you are trying to improve the documentation. I wish you good luck!

But I didn't find the answer to my 2nd question:
"Or what's the difference between output.ptos.forceInternalMechanics, output.ptos.powerInternalMechanics & output.ptos.forceTotal?"
I would really appreciate it if you can help me with this.

dforbush2, 2020-11-20:
Hi @hossein-mehdipour 
We've been getting this question a lot so we definitely will be updating our documentation with these details. See also #461 

The "internalMechanics" fields refer to the power/forces resulting from pto.c and pto.k, if specified, while the forceActuation field refers to the EXTERNAL actuation and total forces on the PTO. For example, in the left-most set of blocks, 

[image]

output.pto.forceActuation will always be zero, as there is no "external" actuation force input. In this case (for instance ./examples/RM3), the heave component of output.pto.forceTotal will be equal to output.pto.forceInternalMechanics, since only pto.k and pto.c provide a force in the heave direction.

However, if an external actuation is prescribed (i.e., a PTO block not in the left-most column is being used) forceActuation will capture this external actuation, forceInternalMechanics will capture forces due to pto.k and pto.c (if specified), and forceTotal = forceActuation + forceInternalMechanics.

Apologies for the confusion I hope this is helpful.

sugarl14, 2020-11-23:
Hey @hossein-mehdipour,

Let me preface by saying I'm not an expert in WEC-Sim, so what I'm about to say will need verification from @dforbush2. Also, bear with me because I like to explain things as simply as possible because I think it helps understand what's really going on. So bear with me!

1) Soooo `output.ptos.forceInternalMechanics`, from my understanding, is essentially the force on the linear PTO. Think of it this way: after an ocean wave hits the OWSC flap, it rotates. The force/torque on the PTO because of that wave is `output.ptos.forceInternalMechanics`. If you're looking at an OWSC, then this would be the torque on the linear PTO (column 5). If you're looking at a point absorber, this would be vertical force (column 3).

1.5) Real quick note: Let's say you're running a simulation on a WEC in WEC-Sim. If you're just using the WEC-Sim input file to run a simulation then you'll have to define pto.k and pto.c. In this case, it would be assumed you're modeling a **linear** PTO. 
However, let's say you want to model a piston-cylinder assembly or desalination system or what-have-you. In this case, you'd use the ptoSimInputFile.m and then you'd specify stuff like piston area, hydraulic accumulator volume, etc. In this case, then it's assumed you're using some sort of **hydraulic** PTO. For now lets keep it simple and go with a linear PTO. With a linear PTO, you'd use one of those two PTO's that @dforbush2 circled in the picture above. For a linear PTO, you won't have a value for `forceActuation` or `forceConstraint(:,5)` (since it's an OWSC, and is free to rotate about this axis).

2) Now let's talk a little bit about output.ptos.powerInternalMechanics.
So we know from school (and google) that Power = Force * velocity.
where,
Power = `output.ptos.powerInternalMechanics`
Force = `output.ptos.forceInternalMechanics`
and
velocity = `output.bodies(1).velocity `

So (again from my understanding),  `output.ptos.powerInternalMechanics`, is the power output from a linear PTO. (Aka, it's the power that the linear PTO can generate after being hit with a wave).

3) As far as `output.ptos.forceTotal`, that is just a sum of all the PTO forces (i.e. `forceActuation`, `forceConstraint`, and `forceInternalMechnics`) But like we talked about in the last sentence of section 1.5, and as @dforbush2 mentioned, since you're using a linear PTO (aka, one of the two PTOs circled in blue), you won't have a value for `forceActuation` or `forceConstraint`, so therefore:

`output.pto.forceTotal' = 'forceInternalMechanics'

**Again, @dforbush2, please let me know if I am wrong/ misleading on anything. I am still learning too!!**

Good luck!

dforbush2, 2020-11-23:
Thanks for your help @sugarl14. This all sounds pretty much right.

It is, however, important to note that for a single DOF PTO, forceInternalMechanics will be non-zero only along a single column, the DOF along which the PTO is acting. The same can be said of forceActuation, in cases where an external actuation force has been prescribed to act along a single DOF at the PTO. However, forceConstraint, forceTotal, and other fields capture forces in all 6 DOF, so in the example @sugarl14 cites in (3), we would expect 

output.pto.forceTotal(:,5) = output.pto.forceInternalMechancics(:,5), 

but the other columns, representing the translational and other rotational directions, will not be equal.

sugarl14, 2020-11-23:
Thank you for the clarification @dforbush2!

closed 2020-12-02 by dforbush2