WEC-Sim/WEC-Sim #448: Degenerate Mass on Sphere

smervosh, 2020-11-02:
Hello,

I am trying to simulate a floating sphere and keep running into this error:

Error using wecSim (line 292)
['BETA/Global Reference Frame/Solver Configuration']: Error evaluating equations at time 0.0.  Stopping simulation.  There may be a
singularity in the solution.  If not, try reducing the step size (either by reducing the fixed step size or by tightening the error
tolerances)
Caused by:
    Error using wecSim (line 292)
    ['BETA/Floating (3DOF)/CONSTRAINT (Planar Joint)']: 'BETA/Floating (3DOF)/CONSTRAINT (Planar Joint)' has a degenerate mass
    distribution on its follower side.

I am running this on Matlab 2020b. Any help with this issue would be greatly appreciated!

Thank you!
[BETA.zip](https://github.com/WEC-Sim/WEC-Sim/files/5475942/BETA.zip)

nathanmtom, 2020-11-04:
@smervosh Thank you for submitting your issue and hope we can help solve your problem. 

After reviewing your Simulink model, it appears that you have a 3DOF constraint connected to a translational PTO. Often in this scenario an additional inertia block needs to be placed between either two PTOs, Constraints, or PTO & Constraint for WEC-Sim to run. This is because the equations of motions between the PTO and Constraint is expecting a mass between the interconnections. Placing a small inertia as shown in the modified .slx should allow your model to get past the singularity at t = 0 s. However, the simulation appears to crash at about 15 mins, but this appears to be a dynamic stability issue rather than a model set-up question. 

Please let us know if you have further questions about this issue. If you are still having issues getting your model to run, but not in regards to the simulation stopping at 0.0 s, please open a new issue.

Regards, 
Nathan
National Renewable Energy Laboratory

[BETA.zip](https://github.com/WEC-Sim/WEC-Sim/files/5488951/BETA.zip)

smervosh, 2020-11-04:
The addition of the inertia fixed the problems I was having with the simulation starting at zero seconds. Thank you!

closed 2020-11-04 by smervosh