WEC-Sim/WEC-Sim #449: Hydrostatic force discrepancy

akeeste, 2020-11-02:
I am adding documentation to the Simulink library and noticed a possible discrepancy in how the hydrostatic restoring force is implemented between the hydrodynamic body and the drag / non-hydro variant subsystems.

The drag and nonhydro bodies only contain a hydrostatic force in the vertical direction. However, the hydrodynamic body contains the vertical net buoyancy force and a moment due to a possible misalignment between the center of gravity and the center of buoyancy.

Why should the drag and nonhydro bodies not contain this hydrostatic moment?

The blocks concerned are:
* Hydrodynamic body: WECSim_Lib/Body Elements/Rigid Body/Hydrodynamic Body/Hydrostatic Restoring Force Calculation/Linear and Nonlinear Restoring Force Variant Subsystem/Linear Hydrostatic Restoring Force 
* Drag body: WECSim_Lib/Body Elements/Rigid Body/Drag Body/Weight and Buoyancy
* Nonhydro body: WECSim_Lib/Body Elements/Rigid Body/Non-Hydro Body/Weight and Buoyancy

dforbush2, 2020-11-11:
Hydrostatic moments exist for surface-piercing bodies. I would argue that a surface-piercing, dynamic body should not be modeled using either the Drag Body or Nonhydrobody methods, as there is clear potential for significant wave interaction. I would suggest that these be modeled as hydrodynamic bodies, which include the noted hydrostatic calculations, and that the Drag Body/non-hydro body implementations are sufficient for their intended uses.

akeeste, 2020-11-16:
@dforbush2 Yes this makes sense. A drag or nonhydrodynamic option should not be used for surface-piercing bodies. I think this moment could still apply to submerged bodies with non-uniform density, maybe due to the placement of internal components. The hydrostatic moment will often be zero for the drag / nonhydro bodies, but I don't think neglecting the hydrostatic moments adds any benefit to the code. Whether the moment is added into the Simulink library or not, I think the documentation (maybe code structure section) should include more information describing the differences between the types of WEC-Sim bodies available.

dforbush2, 2020-11-16:
That's fair. You're correct particular non-uniform bodies will create this moment. I'll see what it would take to add this to the drag bodies implementation.

closed 2020-12-10 by akeeste