WEC-Sim/WEC-Sim #452: Question about yawNonLin implementation

nathanmtom, 2020-11-04:
@dav-og and @dforbush2 I'm curious to get your thoughts about the yawNonLin implementation. I have a model that I would expect the following runs to provide the same outputs:

1. simu.yawNonLin = 1; waves.waveDir = 0;  body(1).initDisp.initAngularDispAxis = [0 0 1]; body(1).initDisp.initAngularDispAngle = -45*(pi/180);

2.  simu.yawNonLin = 1; waves.waveDir = 45;  body(1).initDisp.initAngularDispAxis = [0 0 1]; body(1).initDisp.initAngularDispAngle = 0;

But, currently the response are pretty different and was not able to infer why there might be differences when in theory I'd expect these to be the same.

dforbush2, 2020-11-04:
Hi @nathanmtom 

Just to narrow it down: is this a rigid or a flex body? I'm not sure that this implementation was ever tested with a flexible body and I don't think it will work as-is. And I'm guessing you have BEM coefficients for the body across different orientations bounding the range of yaw positions you expect. 

The set of coefficients used is based off of relative yaw, which is calculated as relYaw = (WaveDir - YawDisplacement * 180/pi) so you're right that this should be giving you identical results. 

If you look in the body(1).hydroForce.fExt field, there should be grids of the excitation coefficients pre-calculated when you run with relative yaw. Can you confirm that these are the same between the two runs? If not there might be a problem with how these grids are pre-calculated when there is an initial displacement specified.

nathanmtom, 2020-11-09:
@dforbush2 Thanks for the follow-up, I do not have any reason to believe there is an error in implementation, but perhaps defining response in the global vs body fixed coordinate systems. 

To answer your questions, the model I am referencing is a rigid body block and I have BEM coefficient data from 0 - 345 deg. 

I'll complete the check you have proposed and get back to you.

nathanmtom, 2020-12-22:
@dforbush2 @dav-og Just wanted to update you that I figured out my initial issue and it was because of neglecting to account for the output.bodies.position vs output.bodies.positionLocal. My apologies for adding this confusion and don't believe there is any issue with the current implementation of the simu.yawNonLin implementation.

closed 2020-12-22 by nathanmtom